mainsail-crew/mainsail #2: Connecting to Klipper randomly appears

orkusmg, 2020-04-25:
During some activities (i.e. QGL) I get Connecting to Klipper notification to pop up, the camera feed does still work though, toolhead is moving on it etc. Console does not get updated anymore, F5 fixes it but console is cleared with every F5 (web page refresh).
[image]

orkusmg, 2020-04-25:
Additional info: it does not happen during print.

meteyou, 2020-04-25:
this is a known bug. the error is reproducible with every command with a execute time over 30-35 seconds. only if the command is pushed by KWC. the response has a timeout.

meteyou, 2020-04-27:
there was a bug in the klipper-api. Pls update klipper!

closed 2020-04-27 by meteyou