mainsail-crew/mainsail #3: End gcode for exhaust fan gets ignored

orkusmg, 2020-04-25:
I had my exhaust fan always configured to run after the print for 3min. With KWC after the print fan starts/ toolhead moves to designated place and fan stops (maybe related to G4 wait command). My end print macro responsible for fan is:
M141 S20 ; Set chamber temp target 20deg - chamber fan on
G4 S180 ; Wait 3min - keeps fan on
M141 S70 ; Set chamber temp target 70deg - chamber fan off

This happened to all prints on new interface, never on octoprint.

meteyou, 2020-04-25:
KWC is only an interface. the print and the gcode processing is only managed by klipper. pls search the error in your klippy.log. Or do you configure your exhaust fan in octoprint? And test the gcode in the console "M141 S20".

meteyou, 2020-04-27:
does it work now?

orkusmg, 2020-04-27:
M141 S20 and M141 S70 work and worked fine in console - this just sets the target for chamber temp which then enables or disables fan. the issue is that fan was starting and switching off immediately.  G4 S180 was being ignorred in between. I have just made fresh install of octopi and klipper + latest interface. Print ongoing - I will update after.

orkusmg, 2020-04-27:
Still the same issue. G4 gets ignored

meteyou, 2020-04-28:
pls don't use octopi! haproxy is starting again, after a reboot and nginx cannot start. i use also for many prints M141 and it works perfekt for me. pls send me your gcode + klippy log. G4 is working if you use the newest klipper API!

meteyou, 2020-04-28:
the gcode is G4 P<milliseconds> and not G4 Sxx. the api has a default timeout of 120seconds! but the timeout is only if you use a marcro-button on the interfacer or send it via console. if it is in the gcode file, klipper itself do it. pls check your [API-Config](https://github.com/Arksine/klipper/tree/work-web_server-20200131/klippy/extras/remote_api) or report this bug arksine.

closed 2020-05-03 by meteyou