mainsail-crew/mainsail #10: Microsoft Edge Temp Graph Display Issue

ArmyAg08, 2020-05-01:
The temperature graph shows a grey background (instead of transparent) as the temperature updates as seen in the picture below. 

[image]

translucentfocus, 2020-05-24:
I know this is a older post, but are you referring to [Microsoft Edge Legacy](https://docs.microsoft.com/en-us/deployedge/microsoft-edge-sysupdate-access-old-edge)?

Using [mainsail-alpha-0.0.12](https://github.com/meteyou/mainsail/tree/v0.0.12) & version 83.0.478.37 (Official build) (64-bit) of Edge (Chromium-based) it works fine 😊

[image]

ArmyAg08, 2020-05-24:
It looks like it might be legacy then. It shows version 44.18362.449.0 for me. I do only see the issue when a heater is active though. I see both are disabled in your screenshot. You might try turning one on to see if you can replicate the behavior.
[image]

translucentfocus, 2020-05-24:
> 
> 
> It looks like it might be legacy then. It shows version 44.18362.449.0 for me. I do only see the issue when a heater is active though. I see both are disabled in your screenshot. You might try turning one on to see if you can replicate the behavior.
> [image]

Yes, that is legacy Edge which uses a [deprecated rendering engine called EdgeHTML](https://blog.mozilla.org/blog/2018/12/06/goodbye-edge/).

Ah, I re-tested and confirmed it works fine when using the latest version of Edge (now Chromium-based using the Blink rendering engine)

I don't know what @meteyou would like to do in this situation, but based on [Microsoft's words here](https://support.microsoft.com/en-us/help/4533505/what-is-microsoft-edge-legacy), I would recommend not trying to support it & allocate development efforts somewhere else.

meteyou, 2020-08-14:
thx for testing different versions... if the problem is solved in the newest version of edge, i dont think i have to fix something there...

closed 2020-08-14 by meteyou