mainsail-crew/mainsail #15: Slicer not recognized

1walkerit, 2020-05-10:
Hi,
I am using IceSL as my slicer but it is not recognized by maisail. Where is it possible to change code to correct it?
Is it possible?
Thank you for help.
[image]

zgrozemag, 2020-05-10:
Check out tripstar76's solution in his printer.lua file, not sure if mainsail reads that metadata same way dwc does though. 
https://github.com/tripstar76/IceSLgit/tree/master/Voron1

meteyou, 2020-05-10:
thanks for the bugreport. mainsail does not read the metadata itself (dwc also for the same reason), because the gcode files themselves are only available on the host system and not on the client system. the metadata are readed by moonraker (klipper-api) and only streamed via api. this extension has to be configure in klipper. (but thanks to arksine) this can be done easily in the printer.cfg. here you can find the examples which are built in so far.

[https://github.com/Arksine/klipper/blob/work-web_server-20200131/config/slicers.cfg](https://github.com/Arksine/klipper/blob/work-web_server-20200131/config/slicers.cfg)

if you have a working regex string, we can implement it, or you make a pull request to arksine's klipper fork.

Arksine, 2020-05-27:
Its actually possible to add support for a new slicer in printer.cfg, however I haven't documented it and it requires some understanding of regular expressions and Jinja2.  I'll look into IceSL and see if I can add support for it

closed 2020-07-16 by meteyou