mainsail-crew/mainsail #22: Nozzle size feature request

ufoDziner, 2020-05-27:
It would be great if the nozzle size could be displayed for 2 items.

1. DASHBOARD indication showing the nozzle size set in the printer.cfg.
2. G-CODE FILES column showing the nozzle size set in each file.

Thanks!

jakep82, 2020-05-29:
Nozzle size in Klipper is only used as a safety check.  Set it to the largest size you normally use, and don't worry about it.  It's not something that needs to be displayed on the dashboard.

ufoDziner, 2020-05-29:
I understand that it's not required in klipper.  Whether it's displayed from the printer.cfg, or stored elsewhere, it's good information to be displayed.  I just thought that using the value from within the printer.cfg would keep everything harmonious.

meteyou, 2020-08-14:
this feature is not interesting at the moment. i will close this FR... maybe there is later a better solution possible...

closed 2020-08-14 by meteyou