mainsail-crew/mainsail #23: unable to load a gcode with a "space" in the filename

manoukianv, 2020-06-07:
Hello,

i have an issue with the filename with an "space" char in the filename :
    1. upload directly the gcode filename with a samba shase => filename with " " is ok
    2. refresh the gcodes files panel => filename is ok in the list
    3. click on it => prompt panel to start print is ok with the space
    4. confirmation print : a error hint "impossible to load gcode"
Not a fatal issue : replacing " " by "_" and all is fine

Thanks a lot for the amazing works !
Vincent

translucentfocus, 2020-06-13:
I'm having a little trouble following your issue 😕

Do you mean that a Samba share **file path** with spaces causes issues?  Or do you mean a **file name** with spaces causes issues if it loaded from a Samba share? Or something else?

If you can explain it better, I'll try to reproduce it also 😊

manoukianv, 2020-06-14:
I mean that a filename with a space in the filename have an issue after file selection in the panel.

meteyou, 2020-06-15:
i never tested a name with spaces, becaus moonraker (the API) replace automatically the spaces to "_". i have to check if it is possible or is the problem in mainsail or moonraker.

meteyou, 2020-06-15:
@manoukianv i ask Arksine why the moonraker replace the spaces. the problem is in klipper. the print start command only execute a "M23" gcode and in gcodes there is no space allowed or cannot be escape.

manoukianv, 2020-06-16:
Hello, i understand. I will check in klipper, it seem to me that it's ok with space, perhaps a mistake from me.

Stephan3, 2020-07-15:
@meteyou 

A workaaround is to rename files during upload.
https://github.com/Stephan3/dwc2-for-klipper/blob/ad8e08088685535d6be82b8aa18c992f46d85643/web_dwc2.py#L992

or during file listing:
https://github.com/Stephan3/dwc2-for-klipper/blob/ad8e08088685535d6be82b8aa18c992f46d85643/web_dwc2.py#L429

meteyou, 2020-07-26:
solved in moonraker...

closed 2020-07-26 by meteyou