mainsail-crew/mainsail #25: Error with IPv6 and Authorization

manoukianv, 2020-06-08:
Hello,

I follow the installation guide in the REDME.md and i have an issue with the "hostname" resolver.
When i call the page with the IP number http://myip/ it works, but with my hostname http://klipper.local/, it doesn't work.

In the log, i have an error 401 and the specified IP in the log is a IPv6 : 
```
2020-06-07 23:22:19,202 [web.py:log_exception()] - 401 GET /websocket (2a01:cb05:xxxx:xxx:xxxxx:49fe:51ef:8bdf): Unauthorized
2020-06-07 23:22:19,217 [web.py:log_request()] - 401 GET /websocket (2a01:cb05:xxxx:xxx:xxxx:49fe:51ef:8bdf) 16.85ms
```

I update the nginx config like that : 
    listen 80 default_server;
    #listen [::]:80 default_server;
And it works now with that config. Nginx seems to reject IPv6, and the log have successful call with the IPv4 address.

I'm bad in networks, not sure if this patch is the good one. But it can be helpful.
Vincent

meteyou, 2020-07-16:
This problem will be solved in Moonraker.

closed 2020-07-16 by meteyou

grigorig, 2022-11-16:
I am still seeing this exact problem. Looks like nothing was solved after all?

meteyou, 2022-11-16:
@grigorig just add your ipv6 range as trusted client in your moonraker.conf or disable ipv6 in nginx (like the example above).

grigorig, 2022-11-17:
The IPv6 prefix may change, so this won't work reliably. Moonraker seems to support user-based authorization, I wonder why this isn't used by default? This seems to be a far superior and more secure solution (your local network usually cannot be trusted - that is a really basic and inadequate security model). I cannot find any documentation on how to actually use user-based authorization from the Mainsail UI, though.

meteyou, 2022-11-17:
> The IPv6 prefix may change, so this won't work reliably. Moonraker seems to support user-based authorization, I wonder why this isn't used by default? This seems to be a far superior and more secure [...]

Because it's not build in. Too many users think, that this feature is to make there printers public!