mainsail-crew/mainsail #30: How to contact you? :>

jin-eld, 2020-07-27:
Hi,

I could not really figure out what the best way to contact you is.

I am hacking a Klipper Linux distro (https://github.com/jin-eld/klipper-linux) which should not be bound to any particular hardware and I was looking for a user interface that would be targeted at Klipper. Your project seems to be doing just that, but I'd like to have a chat before I start integrating anything; and perhaps you'd even be interested in aligning some efforts. Are you by any chance on #klipper IRC on freenode and would you be interested in discussing a few things on the above topic? My nick is Jin^eLD

Kind regards,
Jin

meteyou, 2020-08-01:
hi,

you can join the voron discord and ping me there. my name on the voron discord is "meteTheSailor" at the moment.

here is a invite link: [https://discord.gg/voron](https://discord.gg/voron)

closed 2020-08-04 by meteyou

jin-eld, 2020-08-05:
Hi,

unfortunately I never caught you online, what time zone are you in? Besides, to be honest, I am not keen on registering on Discord and it won't let me write any messages to you with a guest account. If you'd be interested to ping me via mail - it's "jin at mediatmob dot cc"

One thing I'd like to know is if there's a chance that you'd add a Makefile with support for an install target, ideally allowing to set an installation prefix. I'm building the Linux distro witih Yocto (so its a cross build) and while I could probably hack the installation directly in Yoctos bitbake recipes, it would be a great help if the install part was inside  your project, otherwise I'd have to adapt things manually each time you add a file or change something on your side.

I know you do have some scripts available, but these are highly targeted at debian/a particular setup and this is what I wanted to get away from in the first place. The goal of my Klipper distro project is to provide a "ready to go image" where everything comes preinstalled as opposed to starting with a Debian based system and going through  manual installation steps. It should also work on pretty much any device capable of running Linux and not be targeting RasPi specifically. Ideally the user should only have to select his printer from a menu and be able to tune the config, but klipper itself and a web interface should already be there and work out of the box. At least that's what I would like to have as a user.

Currently I am looking at OctoPrint which I am not happy with at all, your project looks a lot more promising especially since it's specifically targeting Klipper.

What I still miss are some system configuration possibilities which would be required for a good user experience, i.e. maybe network settings, webserver password, printer config editing and selection, things like that, was thinking of either making a separate start page for that or look at Octoprint plugins, but maybe parts of your interface could be extended to support something like that as well?

I did not try mainsail yet, that'd be next on my list, but it'd be good to hear if you find any of the above of interest to you, also from the standpoint of becoming the default interface of that Klipper Linux distro. I don't have too much time, so really trying to decide which way to go and what project to chose. And its totally fine if you're not interested in any of it, just let me know and I'll stop bothering you :>