mainsail-crew/mainsail #33: [FR] Possibility to organize MACROS

kakou-fr, 2020-08-09:
Hi,

I just installed Mainsail today, looks like it works, but i as i have a lot of MACROS, i have this rendering.

Could it be possible to organize the MACROS ? 

Select the one we need on each line, and  the order of the MACROS (not only alphabetic ordering) ?

[image]

meteyou, 2020-08-10:
You can hide macros in settings > interface. In backlog there will be a switch for "only during print", "only in idle mode" and "both". More is not planned at the moment...

meteyou, 2020-08-14:
@kakou-fr do you try to hide the unimportant macros on the dashboard?

closed 2020-09-01 by meteyou