mainsail-crew/mainsail #37: Ability to edit config files and macros / programs

changedsoul, 2020-08-14:
Since you closed the other before I could respond, perhaps you could explain how I can get these new added features? I followed the install instruction which seems to get me a version that does not include these changes.
I tried to do a git clone of the master, but now I get no web interface at all.

Try to remember, not everyone knows everything about how to do this stuff. Some need more help than others.

jakep82, 2020-08-14:
The API is in state of flux, and it's not fully merged into Klipper yet.  Harassing the developer like this seems like a great way to get blocked.

changedsoul, 2020-08-14:
Im not trying to, nor do I see how this is harassment? My last post was closed so I was unable to ask further. The response was one of that it assumed I should have known this info and to stop asking such stupid questions.
I would like to use this and or help test it, but I cant even get it to work. I am simply asking for clarification on how to get it to work, since the install instructions do not work.

BlackStump, 2020-08-14:
@changedsoul  you have been at the Moonraker Dev and now at the Mainsail Dev, for a second time.  It is not ready to be used right now, please sit on your hands and wait is that clear enough.
As @jakep82  has said the API is in a state of flux and there is no time frame for this to merged into Klipper.
The current install instructions are out of date and will be updated when it is ready. I would suggest closing your requests as they are not appropriate at this time.

changedsoul, 2020-08-14:
Clear enough, thank you. I can wait.

closed 2020-08-14 by changedsoul

meteyou, 2020-08-14:
thx guys for explain the current state! i am working every free minute for update mainsail to the current API version. there are many changes (important and very nice changes!!!) but i need time to do this! maybe it will be this weekend.