mainsail-crew/mainsail #39: Mainsail title tag for GCODE filename not working properly

translucentfocus, 2020-08-15:
[image]

There's an issue located [somewhere with this code](https://github.com/meteyou/mainsail/blob/9bdcec01cab9bdb2906c99452b3cfa02df93c85d/src/store/getters.js#L253)

My GCODE filename for this test was V350_FilamentCard.gcode if that helps.

I'd love to contribute & fix the code myself, but don't know of a good way to setup a test environment to test changes.

meteyou, 2020-08-17:
is the gcode file in the root directory or in a sub directory?

translucentfocus, 2020-08-18:
It is in the root directory

[image]

closed 2020-09-01 by meteyou