mainsail-crew/mainsail #40: Latest moonraker broke compatibility with Mainsail

asovetov, 2020-08-16:
last worked version moonrakeris is v.08

Moonraker recently received some updates that allow it to function with the official Klipper repo. Unfortunately these updates broke compatibility with Mainsail, as mentioned in moonraker's documentation.

You can however check out an compatible version of Moonraker to get you up and running until Mainsail is updated

robthide37, 2020-08-16:
you must not have read the updated docs . Their is already a new version to fix the webhook change and you no longer use the headless fork

asovetov, 2020-08-16:
> you must not have read the updated docs . Their is already a new version to fix the webhook change and you no longer use the headless fork

disagree. I just did a clean install, with all the latest builds. It didn't work for me. It is possible that the request pool was not included in the extreme build.

meteyou, 2020-08-17:
there are new changes on the klipper side with webhooks and a new dev branch on moonraker. i have to check the current version...

kakou-fr, 2020-08-17:
indeed, i just updated, and nothing works now

translucentfocus, 2020-08-18:
I did a clean install 3 days ago (and haven't updated since).
Mines is working fine with:
Klipper version:  v0.8.0-702-gb490848b
Moonraker version:  v.08-35-ge9cd86f
Mainsail version:  mainsail-beta-0.1.3.zip

Hope this at least helps someone get to a working version

robthide37, 2020-08-19:
i literally had done an update few hrs before that comment so should be able to revert or if anyone wants i can zip up my klipper folder and send over

robthide37, 2020-08-19:
notice as of 17-08-2020:
the newest commits of klipper (aug 15th and newer) will break mainsail/moonraker and therefore it won't work anymore. if you have accidentially updated your klipper installation, roll back to commit b490848 as this is the last known commit which works still fine. 

to roll back, please do the following steps:
1. cd klipper
2. git reset --hard b490848
3. sudo service moonraker restart
4. sudo service klipper restart

klipper/moonraker/mainsail should work fine again now.

robthide37, 2020-08-19:
thats the discord pin

kakou-fr, 2020-08-19:
thks it worked

changedsoul, 2020-08-30:
> I did a clean install 3 days ago (and haven't updated since).
> Mines is working fine with:
> Klipper version: v0.8.0-702-gb490848b
> Moonraker version: v.08-35-ge9cd86f
> Mainsail version: mainsail-beta-0.1.3
> 
> Hope this at least helps someone get to a working version

I apologize for my ignorance, but I am trying to learn git and I have no clue how to find these versions. Can you help me out? I too have updates and things are not broke and wish to get to a working instance once again.

meteyou, 2020-09-01:
fixed with the new version (v0.2.0)

closed 2020-09-01 by meteyou