mainsail-crew/mainsail #43: Cancel_Job - Confused on how to impliment.

changedsoul, 2020-08-16:
I used to have a .gcode file called cancel.gcode that got called when I canceled a print. This no longer works.
I added a section in my config called [CANCEL_PRINT] and got an error saying something about it being already implemented, so I removed it.
I tried renaming my cancel.gcode to cancel_print.gcode, but still no luck.
In my klipper.log, I see where I have tried to pause and cancel:
> action:paused
action:cancel

So I am curious. Is this a bug in mainsail, or should I take this question over to klipper. 
Im sorry if these seem like minor questions.

BlackStump, 2020-08-17:
[did you look at the macro examples in this repo](https://github.com/meteyou/mainsail/blob/master/docs/macros.md)
You may or may not need to adjust to suit your printer

changedsoul, 2020-08-17:
No I had not seen these. Thank you for supplying the link. I’ll implement the examples and give them a try. 
Thanks again.

meteyou, 2020-08-17:
and your gcode file doesn't work anymore, because "RESET_SD" was renamed to "SDCARD_RESET_FILE" in klipper. please RTFM completely, if you update your system...

closed 2020-08-18 by changedsoul