mainsail-crew/mainsail #48: Firefox tab becomes unresponsive when mainsail has been open for a while.

Mechotronic, 2020-08-20:
I've noticed that the longer the Mainsail tab is open the slower it gets. Eventually the tab can't be closed normally, and killing the individual tab process returns the browser back to normal. I'm not sue if this is on other browsers.

v0.8.0-702-gb490848b

Mechotronic, 2020-08-24:
@Torreip I'm on windows 10.

Torreip, 2020-08-26:
@Mechotronic there goes my theorie then :( do you have the problem with chrome or maybe edge? We should probably start collecting logs when that happens.

changedsoul, 2020-08-28:
I can confirm on my arch laptop as well, Firefox has virtually 100% cpu fan blowing nut bar....chromium sits pretty stable around 24% cpu ( that is until I go to the dashboard with the webcam going, then it shoots to 100%).

Perhaps FF even with the webcam not displayed, for example when your on the console tab, is maybe still running the webcam stuff in the background? And chromium only runs it when the camera is actually displayed?

paulhew69, 2020-08-29:
I must say, I have noticed this also. Both my machines run FF, windows10 and Ubuntu 18.04.
Thought it was that I had 2 sessions connected, shut one down, not difference. 
Rebooted the Pi and it was much snappier. I do not run a webcam.

meteyou, 2020-08-29:
I think, it is the temp chart. In the next version, it is completely different. I hope it fix the problem.

meteyou, 2020-09-01:
pls recheck this issue with the new version (V0.2.0). if it is fixed, pls close this issue.

Mechotronic, 2020-09-01:
I followed update.md and now I get an error at /printer/info
```Traceback (most recent call last):
  File "/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/web.py", line 1680, in _execute
    result = self.prepare()
  File "/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/web.py", line 2435, in prepare
    raise HTTPError(self._status_code)
tornado.web.HTTPError: HTTP 404: Not Found```

Mechotronic, 2020-09-01:
I believe moonraker and klipper are fighting now. I will try and hunt it down:
'''Moonraker Log Start...
Git Version: v0.1.0-0-gd59d1e5
================================================================================
2020-09-01 20:57:03,898 [authorization.py:__init__()] - Authorization Configuration Loaded
Auth Enabled: True
Trusted Clients:
192.168.1.0/24
2020-09-01 20:57:03,906 [app.py:register_static_file_handler()] - Registering static file: (/server/files/moonraker.log()) /tmp/moonraker.log
2020-09-01 20:57:03,906 [app.py:register_local_handler()] - Registering local endpoint: (GET POST) /access/api_key
2020-09-01 20:57:03,906 [app.py:register_local_handler()] - Registering local endpoint: (GET) /access/oneshot_token
2020-09-01 20:57:03,909 [app.py:register_local_handler()] - Registering local endpoint: (POST) /printer/print/pause
2020-09-01 20:57:03,909 [app.py:register_local_handler()] - Registering local endpoint: (POST) /printer/print/resume
2020-09-01 20:57:03,909 [app.py:register_local_handler()] - Registering local endpoint: (POST) /printer/print/cancel
2020-09-01 20:57:03,910 [app.py:register_local_handler()] - Registering local endpoint: (POST) /printer/print/start
2020-09-01 20:57:03,910 [app.py:register_local_handler()] - Registering local endpoint: (POST) /printer/restart
2020-09-01 20:57:03,910 [app.py:register_local_handler()] - Registering local endpoint: (POST) /printer/firmware_restart
2020-09-01 20:57:03,911 [moonraker.py:load_plugin()] - Plugin (klippy_apis) loaded
2020-09-01 20:57:03,920 [app.py:register_local_handler()] - Registering local endpoint: (GET) /server/files/list
2020-09-01 20:57:03,920 [app.py:register_local_handler()] - Registering local endpoint: (GET) /server/files/metadata
2020-09-01 20:57:03,921 [app.py:register_local_handler()] - Registering local endpoint: (GET POST DELETE) /server/files/directory
2020-09-01 20:57:03,921 [app.py:register_local_handler()] - Registering local endpoint: (POST) /server/files/move
2020-09-01 20:57:03,921 [app.py:register_local_handler()] - Registering local endpoint: (POST) /server/files/copy
2020-09-01 20:57:03,922 [app.py:register_static_file_handler()] - Registering static file: (/server/files/config/(.*)) /home/pi/klipper_config
2020-09-01 20:57:03,923 [file_manager.py:_update_file_list()] - Updating File List <config>...
2020-09-01 20:57:03,924 [moonraker.py:load_plugin()] - Plugin (file_manager) loaded
2020-09-01 20:57:03,925 [app.py:register_local_handler()] - Registering local endpoint: (POST) /machine/reboot
2020-09-01 20:57:03,925 [app.py:register_local_handler()] - Registering local endpoint: (POST) /machine/shutdown
2020-09-01 20:57:03,926 [moonraker.py:load_plugin()] - Plugin (machine) loaded
2020-09-01 20:57:03,927 [app.py:register_local_handler()] - Registering local endpoint: (GET) /server/temperature_store
2020-09-01 20:57:03,927 [moonraker.py:load_plugin()] - Plugin (temperature_store) loaded
2020-09-01 20:57:03,929 [moonraker.py:load_plugin()] - Plugin (shell_command) loaded
2020-09-01 20:57:03,930 [moonraker.py:start()] - Starting Moonraker on (0.0.0.0, 7125)
2020-09-01 20:57:06,566 [authorization.py:_check_trusted_connection()] - Trusted Connection Detected, IP: 192.168.1.81
2020-09-01 20:57:06,569 [web.py:log_request()] - 101 GET /websocket (192.168.1.81) 5.20ms
2020-09-01 20:57:06,570 [websockets.py:add_websocket()] - New Websocket Added: 3039379312
2020-09-01 20:57:06,575 [websockets.py:dispatch()] - Websocket Request::{"jsonrpc":"2.0","method":"get_directory","params":{"path":"/gcodes"},"id":2576}
2020-09-01 20:57:06,575 [websockets.py:dispatch()] - Websocket Response::{"jsonrpc": "2.0", "error": {"code": 400, "message": "Invalid base path (gcodes)"}, "id": 2576}
2020-09-01 20:57:06,577 [websockets.py:dispatch()] - Websocket Request::{"jsonrpc":"2.0","method":"get_printer_info","params":{},"id":7235}
2020-09-01 20:57:06,577 [websockets.py:dispatch()] - Websocket Response::{"jsonrpc": "2.0", "error": {"code": -32601, "message": "Method not found"}, "id": 7235}
2020-09-01 20:57:06,604 [websockets.py:dispatch()] - Websocket Request::{"jsonrpc":"2.0","method":"get_printer_objects_list","params":{},"id":1744}
2020-09-01 20:57:06,604 [websockets.py:dispatch()] - Websocket Response::{"jsonrpc": "2.0", "error": {"code": -32601, "message": "Method not found"}, "id": 1744}
2020-09-01 20:57:06,605 [websockets.py:dispatch()] - Websocket Request::{"jsonrpc":"2.0","method":"get_directory","params":{"path":"gcodes"},"id":1250}
2020-09-01 20:57:06,605 [websockets.py:dispatch()] - Websocket Response::{"jsonrpc": "2.0", "error": {"code": 400, "message": "Invalid base path (gcodes)"}, "id": 1250}
2020-09-01 20:57:06,606 [websockets.py:dispatch()] - Websocket Request::{"jsonrpc":"2.0","method":"get_directory","params":{"path":"config"},"id":438}
2020-09-01 20:57:06,608 [websockets.py:dispatch()] - Websocket Response::{"jsonrpc": "2.0", "result": {"dirs": [], "files": [{"modified": "Tue Aug 18 03:09:13 2020", "size": 18097, "filename": "printer-20200818_225321.cfg"}, {"modified": "Wed Aug 26 22:31:21 2020", "size": 18094, "filename": "printer - 26AUG2020.cfg"}, {"modified": "Thu Aug 27 23:11:54 2020", "size": 18113, "filename": "printer-20200830_035905.cfg"}, {"modified": "Sat Aug 15 20:38:42 2020", "size": 14816, "filename": "printer-20200815_204507.cfg"}, {"modified": "Wed Aug 26 23:59:14 2020", "size": 18081, "filename": "printer-26AUG2020v2.cfg"}, {"modified": "Sat Aug 15 19:24:41 2020", "size": 14769, "filename": "p [... cut ...]
2020-09-01 20:57:06,609 [websockets.py:dispatch()] - Websocket Request::{"jsonrpc":"2.0","method":"get_directory","params":{"path":"config_examples"},"id":5845}
2020-09-01 20:57:06,610 [websockets.py:dispatch()] - Websocket Response::{"jsonrpc": "2.0", "error": {"code": 400, "message": "Invalid base path (config_examples)"}, "id": 5845}
2020-09-01 20:57:06,610 [websockets.py:dispatch()] - Websocket Request::{"jsonrpc":"2.0","method":"get_printer_gcode_help","params":{},"id":1907}
2020-09-01 20:57:06,611 [websockets.py:dispatch()] - Websocket Response::{"jsonrpc": "2.0", "error": {"code": -32601, "message": "Method not found"}, "id": 1907}'''

meteyou, 2020-09-01:
@Mechotronic do you add the UDS in your /etc/default/klipper?

Mechotronic, 2020-09-01:
-a /tmp/klippy_uds" was missing. I added it and rebooted but still have the same error. moonraker doesn't show the status page though, and still shows the error. I removed it and pulled master and ran the install script. It's still broken. I need to fix moonraker first.

Mechotronic, 2020-09-01:
When i made my changes in nano they didn't save for some reason. It's up and running now. I will test the new temp graph and get back to you. Thanks!

Mechotronic, 2020-09-02:
Worked perfect. Thanks again @meteyou !

closed 2020-09-02 by Mechotronic