mainsail-crew/mainsail #50: Cannot select file to print in G-Code Files

paulhew69, 2020-08-31:
Not a show stopper
When in the G-Code Files tab, If you Right-Click a file, you have the options for Print, Download, Rename
You cannot select print.

meteyou, 2020-09-01:
can you report more informations like browser, browser version, mainsail version, current state of klipper?

paulhew69, 2020-09-01:
Just updated to v0.2.0.
Just right clicked a file, selected print and it asks if I would like to download it.

paulhew69, 2020-09-01:
FireFox 80.0 (64-bit), Ubuntu

meteyou, 2020-09-02:
i cannot reconstruct this error, anyone else have problems with this?

paulhew69, 2020-09-02:
A little video showing the issue
https://www.dropbox.com/s/5u65o1yosrzvfkj/VID_20200902_084230.mp4?dl=0

sebadue, 2020-09-03:
Yes, I can confirm that behavior in 0.21:
https://www.dropbox.com/s/dypkl1tm57xqqh7/MainSail.mp4?dl=0

That seems to be independent from the browser:
Firefox 80.0.1 and Safari 13.1.2 are both affected.

sebadue, 2020-09-03:
Seems I have found the code line to blame:

https://github.com/meteyou/mainsail/blob/25a7e8a33b2ea7d4126ebfa6ed29d7f992edcd6b/src/pages/Files.vue#L152

Calls downloadFile on click.

meteyou, 2020-09-03:
should be fixed in the the current version. pls update and try it again...

sebadue, 2020-09-03:
Thanks, the issue is solved with 0.2.2!

paulhew69, 2020-09-04:
Can confirm issue is fixed - Thank you.

closed 2020-09-04 by paulhew69