mainsail-crew/mainsail #51: Console options - enhancement

paulhew69, 2020-09-01:
Could there be a way of disabling the annoying temp messages in the console, please

meteyou, 2020-09-01:
I don't know what you mean. Can you give me examples?

Mechotronic, 2020-09-02:
He means the B messages that are received for certain commands. 
[image]

meteyou, 2020-09-02:
at the moment, there is no index... its the question, if klipper should not send this values

rafaljot, 2020-09-04:
maybe some regexp patterns to filter out unwanted messages ?

kakou-fr, 2020-09-04:
ability to send a command directly from the console panel in the dashboard will be really well.

jwramseier, 2020-11-17:
[image]
OctoPrint has this capability

[image]

zbrozek, 2020-12-05:
I would also love to have a text entry box to send commands from the main page!

jperrin72, 2020-12-12:
That is a great idea ! I would love to see this feature implemented. All the temperature lines make troubleshooting difficult when you have to go through console output.

meteyou, 2020-12-26:
i added it -> https://github.com/meteyou/mainsail/commit/3c217e1161e093df90f8467067e932a428d44f2a

will be in the next release

closed 2021-01-30 by meteyou