mainsail-crew/mainsail #52: Retract and Extrude buttons do not work

paulhew69, 2020-09-02:
The 'Retract' and 'Extrude' buttons are always greyed out so cannot be used.
I have checked that I have 'min_extrude_temp:' in my printer.cfg

Regards
Paul

[image]

meteyou, 2020-09-02:
update to V0.2.1...

paulhew69, 2020-09-02:
Morning, that was the first I did this morning. Must remember to put more info on an 'issue' for you.
[image]

meteyou, 2020-09-03:
pls send me your printer.cfg... i cannot reconstruct this error...

paulhew69, 2020-09-03:
As requested,

[printer (copy).txt](https://github.com/meteyou/mainsail/files/5170921/printer.copy.txt)

Regards.

meteyou, 2020-09-04:
ok. i cannot find an issue in your config. does the problem still exist with and without min_extrude_temp in V0.2.2? has anyone else this problem?

paulhew69, 2020-09-04:
I am upgrading from octo to Mainsail on my voron. Will let you know.

Regards,
Paul Hewson
Network Specialist

paulhew69, 2020-09-04:
Finished the upgrade, took the temp to 200 and the buttons do not work on my Voron.
Commented out the min_extrude_temp and still no buttons.

paulhew69, 2020-09-04:
I have just worked out what the issue is. You have to keep refreshing the page.
Testing process. 
Nozzle at room temp refresh, buttons greyed out.
Nozzle @ 200, buttons greyed out. refresh the page and the buttons are able to be used.

So to make it work I have to remember to refresh the screen.
Videos to show the issue.
https://www.dropbox.com/sh/fplyck9l1bp2978/AADeVoJf5VUs3SjkqiN1WzoWa?dl=0

Browser : FireFox 80.0.1 (64-bit) running on Ubuntu.
Regards,

Paul

meteyou, 2020-09-12:
should be fixed in the next version.

closed 2020-09-12 by meteyou