mainsail-crew/mainsail #54: Fresh install of MainSail on new SD card fails on nginx

paulhew69, 2020-09-04:
Going to post this here and on Moonraker.
Hardware is a Pi4 connected to 2 SKR1.3's

Installed twice, just incase I did not RTFM correctly the first time!

Please see log file for more info

REgards

Paul.
[Moonraker.txt](https://github.com/meteyou/mainsail/files/5174539/Moonraker.txt)

theopensourcerer, 2020-09-04:
I installed this for the first time last night on a clean raspian and it worked fine. In your log I see:

```bind() to 0.0.0.0:80 failed (98: Address already in use)```

It looks like something is already using port 80. Do you have apache or another web server installed already?

paulhew69, 2020-09-04:
> 
> 
> I installed this for the first time last night on a clean raspian and it worked fine. In your log I see:
> 
> `bind() to 0.0.0.0:80 failed (98: Address already in use)`
> 
> It looks like something is already using port 80. Do you have apache or another web server installed already?

Hi, no.
Fresh install, clean card, freshly etched.
I am literally following the instructions line by line. 
I have installed it 5 times on my other printer this week, thought I would upgrade Octopi to something better on this one!

Regards
Paul

closed 2020-09-04 by paulhew69