mainsail-crew/mainsail #55: 0.2.2: Unable to delete a gcode file

theopensourcerer, 2020-09-05:
Right click on a file in the gcode files screen and choose the delete option. In the javascript console I get a bad request 400 error.  

[image]

I'm using Firefox 80.0.1 on Ubuntu 18.04

theopensourcerer, 2020-09-05:
I should add, before I checked the javascript console, I checked permissions on the directory and files on my RPi. The sdcard dir is 755 pi:pi and the files are all 644 pi:pi.

I wasn't sure if the mainsail process is running under nginx (www-data) so I made the sdcard directory 775 pi:www-data and the files 664 pi:www-data but they were still not able to be deleted.

Hope that helps.

paulhew69, 2020-09-05:
All good here, works fine. Deletes the file, no issues.

th33xitus, 2020-09-05:
Solution: Your filename contains a `%`. Rename that file so that there is no `%` and you can delete that file.

I suspected that the`%` will cause the issue and therefore i quickly tested it myself and could confirm my assumption.

closed 2020-09-05 by meteyou

theopensourcerer, 2020-09-06:
@meteyou Thanks for the swift fix!

commit 458444f0 (2020-09-05): bugfix: delete files with % in filename (fix #55)