mainsail-crew/mainsail #57: New mainsail installation and Failed to start A high performance web server and a reverse proxy server.

tonylabs, 2020-09-06:
I'm trying to install mainsail on a fresh imaged OctoPrint RPi 4.

But got an error when the ./install-moonraker.sh executed.

...
Setting up nginx-full (1.14.2-2+deb10u3) ...
Job for nginx.service failed because the control process exited with error code.
See "systemctl status nginx.service" and "journalctl -xe" for details.
invoke-rc.d: initscript nginx, action "start" failed.
● nginx.service - A high performance web server and a reverse proxy server
   Loaded: loaded (/lib/systemd/system/nginx.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2020-09-06 14:32:41 BST; 50ms ago
     Docs: man:nginx(8)
  Process: 7064 ExecStartPre=/usr/sbin/nginx -t -q -g daemon on; master_process on; (code=exited, status=0/SUCCESS)
  Process: 7065 ExecStart=/usr/sbin/nginx -g daemon on; master_process on; (code=exited, status=1/FAILURE)

Sep 06 14:32:40 octopi nginx[7065]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Sep 06 14:32:40 octopi nginx[7065]: nginx: [emerg] bind() to [::]:80 failed (98: Address already in use)
Sep 06 14:32:40 octopi nginx[7065]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Sep 06 14:32:40 octopi nginx[7065]: nginx: [emerg] bind() to [::]:80 failed (98: Address already in use)
Sep 06 14:32:41 octopi nginx[7065]: nginx: [emerg] bind() to 0.0.0.0:80 failed (98: Address already in use)
Sep 06 14:32:41 octopi nginx[7065]: nginx: [emerg] bind() to [::]:80 failed (98: Address already in use)
Sep 06 14:32:41 octopi nginx[7065]: nginx: [emerg] still could not bind()
Sep 06 14:32:41 octopi systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
Sep 06 14:32:41 octopi systemd[1]: nginx.service: Failed with result 'exit-code'.
Sep 06 14:32:41 octopi systemd[1]: Failed to start A high performance web server and a reverse proxy server.

...

Seems the Nginx install failed. Is that caused by 80 port has been taken by OctoPrint web interface?

Is there a way to fix?

Thanks

theopensourcerer, 2020-09-06:
From the top of the installation instructions:

**Please don't use a octopi image!!!**

https://github.com/meteyou/mainsail/blob/master/docs/installation.md

meteyou, 2020-09-06:
i cannot support too many distros and octopi make every time problems. if you want to use octopi, you have to find your problems for your own, or with other community users... a dont want to support octopi.

closed 2020-09-06 by meteyou

tonylabs, 2020-09-06:
OMG, sorry. I'm blind.