mainsail-crew/mainsail #59: [FR] Webcam stream rotate

Mosher23, 2020-09-07:
Hello,

this is a feature request for camera stream rotation. 
I have tried to flip it in /boot/mainsail.txt as follows `camera_usb_options="-r 1920x1080 -f 30 -rot 180"` but it doesn't work. 

Thank you in advance
[image]

th33xitus, 2020-09-07:
Hmmmm.... turning your webcam is not an option?

Mosher23, 2020-09-07:
@th33xitus no, its not an option.

trevjonez, 2020-09-07:
if you are using ffmpeg to produce the stream maybe [this SU answer](https://superuser.com/a/578329) has what you need to fix it.

or if you are using mjpg-streamer you need to add the `-rot` argument. [more on that here](https://community.octoprint.org/t/available-mjpg-streamer-configuration-options/1106)

While there is probably some sort of css that could flip it over in the receiving browser that mainsail could support, I'd urge you to push the problem upstream to resolve it as close to the webcam as possible. that way your fix isn't tied to any specific consumption format.

lixxbox, 2020-09-08:
what is `/boot/mainsail.txt`? which cam do you use? which OS?
some more infos would be helpful.

`-rot` seems to be only available for raspicams, not usbcams.

Mosher23, 2020-09-08:
should work also for usb webcams [here look](https://github.com/jacksonliam/mjpg-streamer/tree/master/mjpg-streamer-experimental/plugins/input_uvc)

th33xitus, 2020-09-08:
note:
> Optional parameters **(may not be supported by all cameras):**

have you tried the other parameters?
flipping vertically once, and then horizontally also once would result in a 180° rotation.

lixxbox, 2020-09-08:
if it should work, i personally see no reason to integrate this feature into mainsail.
and without further information nobody can help you. at least not me. 🙃

dieideeistgut, 2020-09-08:
Rotating the Stream is depending on the Webcam for a lot of tools (ustreamer, mjpegstreamer, ...). I have implemented a rotate and scale function here: https://github.com/dieideeistgut/mainsail/tree/feature/webcam-rotate (it's quite old). But one may be able to copy from there.

meteyou, 2020-09-12:
i merged today a PR with webcam flip. with this you can rotate your cam.

closed 2020-09-12 by meteyou

gayuha, 2021-02-05:
What about quarter turn rotation? My camera is mounted on a wall, so flipping is not enough. `-rot 90` doesn't work for me.

gcormier, 2021-02-28:
Would also appreciate 90 degree rotation.

gcormier, 2021-02-28:
I found the commit b216ead36d3895ae649694a08eb2de35e5d9a7a7 , and if I manually edit the JSON file, it does rotate it. 

Perhaps a new FR to expose this in the config somehow?

[image]

meteyou, 2021-02-28:
do you see the overflow of the image? thats the reason, why it is deactivated in mainsail. pls try to rotate your webcan in mjpeg streamer 90°.

gcormier, 2021-02-28:
How do I rotate it in mjpeg?