mainsail-crew/mainsail #60: Full Screen web app on iOS.

LazaroFilm, 2020-09-07:
It would be great if Mainsail was optimized to work as an iOS web app. that would allow for fullscreen option and a pretty icon and all. 
http://www.onlywebpro.com/2015/07/19/optimizing-full-screen-mobile-web-app-for-ios/

meteyou, 2020-09-13:
thanks a lot for the tip. i will do it once and therefore i leave this issue open as a reminder.

marqueeeeeee, 2020-09-17:
I was about to open an issue for this, i think PWA is the way achieve this. I’ll try to tinker with the source and file a PR if i am able to do it.

LazaroFilm, 2020-09-20:
I'd love to help too. But I'm still learning JavaCript and React right now. Not quite there yet... Some day maybe... Looking forward to it. In the meantime I'm using my old Octoprint Webapp link that was linking to my printer's IP, so I have an Octoprint icon on the phone, but when I tap it it opens Mainsail in full screen.

meteyou, 2020-09-21:
feel free to make a PR...

translucentfocus, 2020-09-22:
I have a PR in for [the icon issue](https://github.com/meteyou/mainsail/pull/80), but not the iOS web app issue.

translucentfocus, 2020-09-22:
> I'd love to help too. But I'm still learning JavaCript and React right now. Not quite there yet... Some day maybe... Looking forward to it. In the meantime I'm using my old Octoprint Webapp link [...]

Are you willing to test? I don't have an iOS device but have made a patch located here:  https://github.com/translucentfocus/mainsail/releases/tag/v1.0

If for some reason it doesn't break mainsail, it's easy to revert by running
`cd ~/mainsail`
`wget -q -O mainsail.zip https://github.com/meteyou/mainsail/releases/download/v0.2.2/mainsail-beta-0.2.2.zip && unzip mainsail.zip && rm mainsail.zip`

LazaroFilm, 2020-09-23:
> Are you willing to test? 

Yes! I installed your branch but don't see a difference in the "Add to Home Screen" behavior. 
I made a [video](https://youtu.be/Ub6AcyskINw) showing it

translucentfocus, 2020-09-24:
> 
> 
> > Are you willing to test?
> 
> Yes! I installed your branch but don't see a difference in the "Add to Home Screen" behavior.
> I made a [video](https://youtu.be/Ub6AcyskINw) showing it

Awesome! Thanks for trying it... sorry it didn't work... I'm working on a new one to test, but there's still some issues with with it right now.

closed 2020-10-04 by meteyou

LazaroFilm, 2020-10-22:
This is now 100% working since #80 both for the home screen icon and the full screen mode! Thanks!