mainsail-crew/mainsail #61: Selectable temps on the graph

paulhew69, 2020-09-07:
Am I wrong in thinking in the earlier versions of MainSail, you could select what temps were displayed on the graph, by clicking them.
Does not seem to work anymore if it was there. Could it be re-implemented again, please?

meteyou, 2020-09-13:
it will come again. it's just a bit more complicated with the new temp data synchronization.

witem, 2020-09-19:
@meteyou maybe you need helps with it? I can try to implement it

meteyou, 2020-09-19:
@witem thx for your help. i already solved it on my locally branch, but its only the first try. i have to test it a little bit more.

closed 2020-09-19 by meteyou

witem, 2020-09-19:
@meteyou okay, you have some roadmap xto improve this project? I can try to help with some other features or bugs

commit 75d742ac (2020-09-19): bugfix: hide temps in tempchart (fixed #61)