mainsail-crew/mainsail #63: Issues with uploading gcode to Mainsail

MalsakerLures, 2020-09-10:
Hi there,

The mainsail interface looks awesome, just intalled it on a raspeberry Pi 3 following the install instructions.
I do not have any printer connected to the Pi just yet as it is at another location and im not sure if this is required for me to be able to upload the gcode files or not? I guess this should not matter, or?

Nor can i create a new directory, i get the error: Invalid base path (gcode)

meteyou, 2020-09-10:
without klipper, mainsail only run in "emergency mode" and gcode-files only work correctly with a running klipper instance.

MalsakerLures, 2020-09-10:
Ahh, ok, well klipper is installed and printer.cfg is read aswell are the Macros. But i guess i the need the mcu-connected aswell before things work as per expectations?

Skickat från min iPhone

MalsakerLures, 2020-09-10:
> When Moonraker (the backend) initially starts it does not know where the gcode path is located. It retreives that information from Klipper on the initial connection after Klippy enters the "ready" state. Once Moonraker has that path it will retain it regardless of Klipper's state until Moonraker is restated.
>
> Thus you will need to set up a basic working configuration for Klipper with [virtual_sdcard] included before Moonraker can serve the gcode path.

And that should already be in my printer.cfg from my existing klipper install that i have copied and pasted into the new printer.cfg for mainsail.

Need to check

Skickat från min iPhone

Arksine, 2020-09-10:
>But i guess i the need the mcu-connected aswell before things work as per expectations?

Right, you will need an MCU connected so Klipper can enter "ready".  Once Klipper has successfully started you will be able to access the gcode path.

closed 2020-09-12 by meteyou