mainsail-crew/mainsail #66: Heightmaps not being saved within MainSail

paulhew69, 2020-09-12:
Just fitted a BLTouch to my printer. Lots of tests Homing and QGL's
Went to the HeightMap tool, clicked 'Calibrate' let it run a mesh. It displayed, Clicked 'Save' gave it a name, clicked save.
Changed the build surface, ran another 'Calibrate' Saved under a different name.
Went to load the other saved mesh and it is not in the list, nor is the latter one.
Checked printer.cfg and I only have 2 meshmaps in there, the 2 new ones and 'default' are not there either.

meteyou, 2020-09-12:
you have to execute "save_config" for save it in the printer.cfg

closed 2020-09-12 by meteyou

paulhew69, 2020-09-12:
So i created the heightmaps from the console BLT100 and BLT105, ran save_config, because it reminds you too, which is handy.
(could you enable a reminder based on change if printer.cfg or anything else that might require a reboot has changed - suggestion)

Back into the heightmap page clicked the profiles and they were not there. Checked printer.cfg and they were there.
Had to refresh the page for MainSail to look for the new profiles, and my new BLT ones were there.

Will you be fixing the refresh issue, because it happens a lot.
If you reboot the printer, there is a spinning wheel on the button, unless you refresh that page.
Same with my other issue with the Extrude and retract buttons, unless you refresh the page it will not give the correct status as they do not update in realtime.