mainsail-crew/mainsail #73: 64bit

AlessandroEmm, 2020-09-15:
Sorry, this is more of a question:

Is there a reason you recommend the 32bit raspy os in the installation?

Thanks a lot for what you've created!

RaymondHimle, 2020-09-15:
32bit is recommended as there is no official Raspberry Pi OS in 64bit at this time.

meteyou, 2020-09-16:
as @RaymondHimle said. the 32bit version is the current recommented version from Raspberry Pi OS

[https://www.raspberrypi.org/downloads/raspberry-pi-os/](https://www.raspberrypi.org/downloads/raspberry-pi-os/)

closed 2020-09-19 by meteyou

TimoMF, 2022-05-20:
The official final 64-bit version of the Raspberry Pi OS has been available since the beginning of the year.

Are you already considering releasing a 64-bit version of MainsailOS?

For those who have the 4 or 8GB version, there might be small benefits.

Greetings Timo

https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-64-bit

meteyou, 2022-05-20:
PPS Uster the mainsailOS issue tracker for mainsailOS questions...