mainsail-crew/mainsail #74: Height map calibration

krpepe, 2020-09-15:
Hello.

Thank you for the wonderful GUI for klipper. I have one question about calibrate in height map. I have a custom probe for my Z axis which requires a retract prior to probing. If I run my Bed mesh macro in console everything is fine but if I use calibrate in height map it tells me the probe is triggered. What code is the calibrate button using and can it be modified?

Thank you in advance,
Kevin

meteyou, 2020-09-19:
you can use a macro with rename_existing attribute like this:

`
#
#	BED_MESH_CALIBRATE - Bed Mesh Leveling
#

[gcode_macro BED_MESH_CALIBRATE]
rename_existing: BASE_BED_MESH_CALIBRATE
gcode:
	BED_MESH_CLEAR
	QUAD_GANTRY_LEVEL
	G1 X125 Y125 Z5 F6000
	BASE_BED_MESH_CALIBRATE
`

krpepe, 2020-09-19:
@meteyou Thank you, I will give that a try!

krpepe, 2020-09-19:
@meteyou Unfortunately that didn't correct it.

meteyou, 2020-09-19:
then I cannot help you. the button in heightmap only execute "BED_MESH_CALIBRATE". with the macro example, you can modify this command. if it doesn't work, you have a klipper problem, or a problem with your custom probe!

krpepe, 2020-09-19:
@meteyou Ok thank you! I know it isn't my probe because if I use the bed mesh macro listed on the on the dashboard it works flawlessly. The only time I have an issue is with calibrate under height map and z-tilt on the dashboard as well. My z-tilt macro works from the dashboard as well. Ill play around with it and if worse comes to worst I go back to octoprint. Thank you for your Time!! Have a great day!

meteyou, 2020-09-19:
But why you cannot use rename_existing with your macro?

krpepe, 2020-09-19:
@meteyou I tried it and it didn't correct the issue.

meteyou, 2020-09-19:
Thx for no information! What do you try correctly? What's is your dashboard macro? What is the correct issue? No informations = no help!

krpepe, 2020-09-19:
@meteyou this is what I did
[gcode_macro BED_MESH_CALIBRATE]
rename_existing: BASE_BED_MESH_CALIBRATE
gcode:
   pin_release
	BED_MESH_CLEAR
	G4 P500
	BASE_BED_MESH_CALIBRATE
	pin_retract

meteyou, 2020-09-19:
And how look your dashboard macro? Where is you log file?

krpepe, 2020-09-19:
@meteyou the information is as follows
[image]

meteyou, 2020-09-19:
Can you show me the content of this macro! I know how a dashboard macro button looks like! Pls send me your config + a klipper log with the error!

krpepe, 2020-09-19:
@meteyou If i use my bed mesh macro on the dashboard it does the pin_retract macro if I use Calibrate under the height map it ignores it and give me the probe triggered error. Same occurs with Z-TILT by the HOME ALL button. If I use that button it it gives the probe triggered error if I use my Macro it works fine.

here is my entire config

 I will post the klipper log shortly

[stepper_x]
step_pin: ar54
dir_pin: !ar55
#enable_pin: multi_pin:controller_fan
enable_pin: !ar38
step_distance: .0125
endstop_pin: ^!ar3
position_min: 0
position_endstop: 0
position_max: 200
homing_speed: 30
homing_retract_dist: 2
second_homing_speed: 1
homing_positive_dir: false

[tmc2130 stepper_x]
cs_pin: ar53 
microsteps: 16
interpolate: True
run_current: .8
hold_current: .1
sense_resistor: 0.110
stealthchop_threshold: 200
#driver_IHOLDDELAY: 8
#driver_TPOWERDOWN: 0
#driver_TBL: 1
#driver_TOFF: 4
#driver_HEND: 7
#driver_HSTRT: 0
#driver_PWM_AUTOSCALE: True
#driver_PWM_FREQ: 1
#driver_PWM_GRAD: 4
#driver_PWM_AMPL: 128
#driver_SGT: 0
#diag1_pin:

[stepper_y]
step_pin: ar60
dir_pin: !ar61
enable_pin: !ar56
step_distance: .0125
endstop_pin: ^!ar14
position_endstop: 0
position_min: 0
position_max: 200
homing_speed: 30
homing_retract_dist: 2
second_homing_speed: 1
homing_positive_dir: false

[tmc2130 stepper_y]
cs_pin: ar64
microsteps: 16
interpolate: True
run_current: .8
hold_current: .1
sense_resistor: 0.110
stealthchop_threshold: 200
#driver_IHOLDDELAY: 8
#driver_TPOWERDOWN: 0
#driver_TBL: 1
#driver_TOFF: 4
#driver_HEND: 7
#driver_HSTRT: 0
[... 383 lines cut ...]
#*# control = pid
#*#
#*# [bed_mesh default]
#*# version = 1
#*# points =
#*# 	-0.047585, -0.027585, -0.010085, -0.002585, -0.002585, -0.010085, -0.022585, -0.030085, -0.042585, -0.052585
#*# 	-0.047585, -0.030085, -0.015085, -0.005085, -0.000085, -0.005085, -0.017585, -0.022585, -0.037585, -0.050085
#*# 	-0.032585, -0.020085, -0.010085, 0.002415, 0.004915, 0.002415, -0.007585, -0.020085, -0.030085, -0.035085
#*# 	-0.027585, -0.010085, 0.002415, 0.012415, 0.017415, 0.017415, 0.007415, -0.002585, -0.015085, -0.025085
#*# 	-0.002585, 0.004915, 0.019915, 0.034915, 0.029915, 0.024915, 0.017415, 0.004915, -0.012585, -0.020085
#*# 	0.004915, 0.019915, 0.032415, 0.042415, 0.044915, 0.042415, 0.032415, 0.024915, 0.009915, -0.002585
#*# 	0.017415, 0.029915, 0.047415, 0.054915, 0.054915, 0.057415, 0.044915, 0.034915, 0.022415, 0.009915
#*# 	0.009915, 0.024915, 0.039915, 0.052415, 0.059915, 0.057415, 0.052415, 0.039915, 0.032415, 0.014915
#*# 	-0.005085, 0.017415, 0.029915, 0.044915, 0.059915, 0.062415, 0.057415, 0.054915, 0.042415, 0.034915
#*# 	-0.055085, -0.027585, -0.002585, 0.017415, 0.029915, 0.042415, 0.049915, 0.044915, 0.044915, 0.044915
#*# x_count = 10
#*# y_count = 10
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = bicubic
#*# tension = 0.1
#*# min_x = 25.0
#*# max_x = 199.9625
#*# min_y = 2.00000000002
#*# max_y = 198.925

krpepe, 2020-09-19:
[image]

[klippy.zip](https://github.com/meteyou/mainsail/files/5250104/klippy.zip)

krpepe, 2020-09-19:
Is there anything else you need?

krpepe, 2020-09-19:
@meteyou This is what I replaced my gcode_macro bed_mesh with and it didn't work

[gcode_macro BED_MESH_CALIBRATE]
rename_existing: BASE_BED_MESH_CALIBRATE
gcode:
    pin_release
	BED_MESH_CLEAR
	G4 P500
	BASE_BED_MESH_CALIBRATE
	pin_retract

meteyou, 2020-09-19:
i don't see this in your log, but this should work. maybe you forget the tab before each gcode line?

but... the main problem is, that i cannot find a line with "rename" in your logfiles, so your config is not correct. but this is all a klipper config issue and not a mainsail issue.

krpepe, 2020-09-19:
@meteyou 
[image]
[klippy.log](https://github.com/meteyou/mainsail/files/5250394/klippy.log)

krpepe, 2020-09-19:
@meteyou As you can see from the screen shot it would appear to be there  but in klippy log it isn't. Which config needs to be changed? As you can see I am changing it with the editor in mainsail and it modifies the one in pi\klipper_config\printer.cfg. does the original need to be modified as well?
[image]

[image]
[klippy.log](https://github.com/meteyou/mainsail/files/5250398/klippy.log)

meteyou, 2020-09-19:
work other changes? is it your correct printer.cfg? check your /etc/default/klipper, if you set it to /home/pi/klipper_config/printer.cfg.

krpepe, 2020-09-19:
That was it! Thank you so much! I am sorry for my mistake! Thank you for taking the time to help me figure it out!
Have a great weekend!

Kevin

closed 2020-09-19 by meteyou

meteyou, 2020-09-19:
@krpepe i added this macro overwrite in my docs now -> [https://github.com/meteyou/mainsail/blob/master/docs/macros.md#customize-klipper-default-commands](https://github.com/meteyou/mainsail/blob/master/docs/macros.md#customize-klipper-default-commands)