mainsail-crew/mainsail #76: Cannot access GCODE FILES repository

lanman1, 2020-09-17:
I'm running Klipper/Mainsail installed with the kiauth script on a Raspi4
I cannot upload gcode to or use the UI to create a Dir in the GCODE FILES Dir.  In the printer.cfg I have 

[virtual_sdcard]
path: /home/pi/sdcard

And that dir exists.  The mookraker.log shows this:

2020-09-17 17:22:01,358 [websockets.py:dispatch()] - Websocket Request::{"jsonrpc":"2.0","method":"server.files.get_directory","params":{"path":"gcodes"},"id":649}
2020-09-17 17:22:01,359 [websockets.py:dispatch()] - Websocket Response::{"jsonrpc": "2.0", "error": {"code": 400, "message": "Invalid base path (gcodes)"}, "id": 649}
2020-09-17 17:22:03,555 [websockets.py:dispatch()] - Websocket Request::{"jsonrpc":"2.0","method":"server.files.get_directory","params":{"path":"gcodes"},"id":9737}
2020-09-17 17:22:03,556 [websockets.py:dispatch()] - Websocket Response::{"jsonrpc": "2.0", "error": {"code": 400, "message": "Invalid base path (gcodes)"}, "id": 9737}

meteyou, 2020-09-18:
Is klipper ready?

lanman1, 2020-09-19:
Klippy-Status is 'shutdown' because I am testing this on a mainboard that doesn't have thermisors connected so I have a "MCU 'mcu' shutdown: ADC out of range" message.  Would this prevent seeing the virtual sdcard dir?

meteyou, 2020-09-19:
https://github.com/meteyou/mainsail/issues/63

closed 2020-09-19 by meteyou