mainsail-crew/mainsail #77: Consider alternate location for gui.json

tumbleweedlabs, 2020-09-18:
Currently, changes made to the GUI interface of mainsail are captured in a file called gui.json and stored in the virtual SD location /home/username/sdcard. It appears that the Mainsail interface hides this file from the user (a good thing) but the file does show up if the virtual SD card is mounted as an SMB share which creates a potentially undesirable situation. This file can easily be deleted when managing files (clearing out old gcode files). I'm writing to ask for consideration that this file be stored in an alternate location. Screenshots here: https://imgur.com/a/hq16Cf1

meteyou, 2020-09-18:
At the beginning of moonraker, I had only access to the virtual sdcard directory. Since on of the last versions, I have access to the config files. If I have to much time, I will move it to the config directory, but it's not important for me, because SMB is a user created issue.

You can also hide files in your SMB config, if this is a big problem for you. 

To delete this file is no big problem. You only lose your hidden macro settings and webcam settings.

meteyou, 2020-09-27:
i fixed this issue and moved the gui.json to klipper_config.

closed 2020-09-27 by meteyou

tumbleweedlabs, 2020-09-29:
Awesome. Thank you!