mainsail-crew/mainsail #79: Remove unused Javascript Files

translucentfocus, 2020-09-21:
**Describe the bug:**\
Lighthouse reports that there are two js files (one that's rather large) that are unused.

**Expected behavior:**\
Unused javascript files should be removed

**Screenshots:**\
[image]

**Logfiles:**\
I don't believe logs are necessary for this issue.  However, version info may be important.  I am running `v0.2.2 - v0.8.0-779-g91979756`

meteyou, 2020-09-21:
do you know, that mainsail is a vue.js project with the framework vuetify? the app.5dc990c3.js is the main code of mainsail itself. chunk-vendors.js is the most part of the framework. if you think a seo tool handle a webapp, give it a try and delete this files ;-)

translucentfocus, 2020-09-21:
Sorry for wasting your time!  I am aware it's a Vue app with Vuetify (and it's awesome btw).  I guess deleting the main web app would make it faster 🤣

closed 2020-09-21 by translucentfocus