mainsail-crew/mainsail #81: refreshing the webcam page returns 502 Bad Gateway

LazaroFilm, 2020-09-22:
**Describe the bug:**\
On the Webcam page, hit refresh. Or, going to the address `myprinter.local/webcam/` shows a `502 Bad Gateway
nginx/1.14.2`

**Expected behavior:**\
See the webcam page.

**Screenshots:**\
[image]

**Logfiles:**\
Please upload your Logfiles here or upload them to [pastebin.com](http://pastebin.com) and paste your log links here.
- [klippy (1).log](https://github.com/meteyou/mainsail/files/5263829/klippy.1.log)
- [moonraker (1).log](https://github.com/meteyou/mainsail/files/5263835/moonraker.1.log)

meteyou, 2020-09-23:
do you install & configure mjpgstreamer or something else for your webcam?

LazaroFilm, 2020-09-23:
I followed [these instructions](https://3dp.tumbleweedlabs.com/firmware/klipper-firmware/adding-webcam-support-to-mainsail) (with some tweaks to get raspicam working)

meteyou, 2020-09-23:
thx for the report. mainsail route to /webcam and nginx have a redirect for /webcam to port 8080 (mjpgstreamer). this is the problem, why you are redirect after a refresh.

your error 502 comes from your mjpgstreamer setup. this is no mainsail issue.

commit e6c19065 (2020-09-23): bugfix: navipoint webcam and nginx redirect use the same url (fix #81)
closed 2020-09-23 by meteyou