mainsail-crew/mainsail #86: Problem buring installing klippy?

farkasn, 2020-09-28:
Hi. i have an issue during the installion and I don't know what I'm doing wrong... :(

I have installed rapbian, Mainsail, Moonraker and tried to run it on my local ip

When I open  it (http://192.168.0.220/) than it just connecting and nothing happens.

First I opened the printer info and it looks like this:

`Traceback (most recent call last):
  File "/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/web.py", line 1680, in _execute
    result = self.prepare()
  File "/home/pi/moonraker/moonraker/authorization.py", line 188, in prepare
    raise tornado.web.HTTPError(401, "Unauthorized")
tornado.web.HTTPError: HTTP 401: Unauthorized (Unauthorized)`

I think this is not normal. :)
Should I add my loacl ip to the trusted_clients? (it is not working ;))
My klippy.log looks like this:
`pi@raspberrypi:/tmp $ nano klippy.log
  GNU nano 3.2                                                             klippy.log

cs_pin = PA3
sclk_pin = PA1
sid_pin = PC1
encoder_pins = ^PD2, ^PD3
click_pin = ^!PC0
=======================
Extruder max_extrude_ratio=0.266081
Starting serial connect
webhooks client 1969467264: New connection
webhooks client 1969467264: Client info {'program': 'Moonraker', 'version': 'v0.1.0-33-ga578199'}
Loaded MCU 'mcu' 80 commands (v0.8.0-530-g1d2746f1-20200521_170204-octopi / gcc: (GCC) 5.4.0 binutils: (GNU Binutils) 2.26.20160125)
MCU 'mcu' config: RECEIVE_WINDOW=192 BUS_PINS_spi=PB6,PB5,PB7 SERIAL_BAUD=250000 STEP_DELAY=-1 ADC_MAX=1023 PWM_MAX=255 RESERVE_PINS_serial=PD0,PD1 BUS_PINS$
Protocol error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 151, in _connect
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 573, in _connect
    self._send_config(None)
  File "/home/pi/klipper/klippy/mcu.py", line 512, in _send_config
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 251, in _build_config
    cq=cmd_queue)
  File "/home/pi/klipper/klippy/mcu.py", line 669, in lookup_command
    return CommandWrapper(self._serial, msgformat, cq)
  File "/home/pi/klipper/klippy/mcu.py", line 386, in __init__
    self._cmd = serial.get_msgparser().lookup_command(msgformat)
  File "/home/pi/klipper/klippy/msgproto.py", line 303, in lookup_command
    msgformat, mp.msgformat))
error: Command format mismatch: schedule_soft_pwm_out oid=%c clock=%u on_ticks=%u off_ticks=%u vs schedule_soft_pwm_out oid=%c clock=%u on_ticks=%u
No build file /home/pi/klipper/klippy/../.config
No build file /home/pi/klipper/klippy/../out/klipper.dict
No build file /home/pi/klipper/klippy/../out/klipper.elf`

th33xitus, 2020-09-29:
> Should I add my loacl ip to the trusted_clients? (it is not working ;))

Yes. But why it isn't working?

farkasn, 2020-09-29:
I have added this line:
trusted_clients:
    192.168.0.220/24

sudo service klipper restart
sudo service moonraker restart

And when I open my printer info (http://192.168.0.220/printer/info) I get 502 Bad Gateway and when I open http://192.168.0.220 still get the connecting screen... :(

[image]

th33xitus, 2020-09-29:
Add `127.0.0.1` to the trusted clients as well.

farkasn, 2020-09-29:
Still connecting. The end of my klippy.log looks like this. Sorry... :(

`Extruder max_extrude_ratio=0.266081
Starting serial connect
Loaded MCU 'mcu' 80 commands (v0.8.0-530-g1d2746f1-20200521_170204-octopi / gcc: (GCC) 5.4.0 binutils: (GNU Binutils) 2.26.20160125)
MCU 'mcu' config: RECEIVE_WINDOW=192 BUS_PINS_spi=PB6,PB5,PB7 SERIAL_BAUD=250000 STEP_DELAY=-1 ADC_MAX=1023 PWM_MAX=255 RESERVE_PINS_serial=PD0,PD1 BUS_PINS$
Protocol error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 151, in _connect
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 573, in _connect
    self._send_config(None)
  File "/home/pi/klipper/klippy/mcu.py", line 512, in _send_config
    cb()
  File "/home/pi/klipper/klippy/mcu.py", line 251, in _build_config
    cq=cmd_queue)
  File "/home/pi/klipper/klippy/mcu.py", line 669, in lookup_command
    return CommandWrapper(self._serial, msgformat, cq)
  File "/home/pi/klipper/klippy/mcu.py", line 386, in __init__
    self._cmd = serial.get_msgparser().lookup_command(msgformat)
  File "/home/pi/klipper/klippy/msgproto.py", line 303, in lookup_command
    msgformat, mp.msgformat))
error: Command format mismatch: schedule_soft_pwm_out oid=%c clock=%u on_ticks=%u off_ticks=%u vs schedule_soft_pwm_out oid=%c clock=%u on_ticks=%u
No build file /home/pi/klipper/klippy/../.config
No build file /home/pi/klipper/klippy/../out/klipper.dict
No build file /home/pi/klipper/klippy/../out/klipper.elf
`

meteyou, 2020-10-01:
`trusted_clients: 192.168.0.220/24`is not correct. is a mix between ip address and network. your network is `192.168.0.0` and netmask is `255.255.255.0`. this is 24. so trusted clients should be `192.168.0.0/24` for unlock all ip addresses between 192.168.0.1 and 192.168.0.254.

`/dev/serial/by-id/usb-FTDI_FT232R_USB_UART_A107BRGS-if00-port0` looks like a not flashed board. you have to install/flash klipper on your board like a normal klipper installation.

closed 2020-10-04 by meteyou