mainsail-crew/mainsail #87: Setting chamber temperature in mainsail

khumarahn, 2020-09-30:
Hi, I set up mainsail today from the current repo. My printer has a chamber heater, defined in printer.cfg with
```
[heater_generic chamber_heater]
gcode_id: C
...
```

When I attempt to set its temperature in the web interface, I get an error
```
!! Malformed command 'SET_HEATER_TEMPERATURE HEATER=heater_generic chamber_heater TARGET=50'
```
I guess the command should be
```
SET_HEATER_TEMPERATURE HEATER=chamber_heater TARGET=50
```

meteyou, 2020-10-01:
ahh. thx for the bugreport. i only use/tested extruders, heater_bed and temperature_fans but no heater_generic. i will update this.

khumarahn, 2020-10-15:
Hi. I get the same error with version 0.23. (thanks for the new version!!)

aka13-404, 2020-10-17:
Can confirm it still happening, web interface looks strange as well. See attached screenshots.
[image]
[image]
[image]

khumarahn, 2020-10-18:
is this with 0.24?

@meteyou maybe re-open this issue?

meteyou, 2020-10-18:
pls test the attached build. i test it only with a thermistor...

[mainsail-beta-nightly-heater_generic.zip](https://github.com/meteyou/mainsail/files/5397563/mainsail-beta-nightly-heater_generic.zip)

khumarahn, 2020-10-18:
Thanks! I installed everything with kiauh, as a side effect I don't know how to do anything manually. What is the easiest way to try this build?

khumarahn, 2020-10-18:
Say, is it enough to override `/home/pi/mainsail' with the contents of the archive? Does mainsail save any settings in `/home/pi/mainsail'?

meteyou, 2020-10-18:
`
cd ~/mainsail
rm -R ./*
wget -q -O mainsail.zip https://github.com/meteyou/mainsail/files/5397563/mainsail-beta-nightly-heater_generic.zip && unzip mainsail.zip && rm mainsail.zip
`
and then open mainsail and refresh with CTRL+SHIFT+R for reload without cache.

khumarahn, 2020-10-18:
It looks like it works. So far I updated mainsail and changed chamber temperature in the middle of the print, and it looks like it worked. I will try again soon when the print finishes.

meteyou, 2020-10-18:
thx. if it works without problems, pls close this issue. this update should be in the next version.

aka13-404, 2020-10-18:
Works great, no issues, the view as well as temperature setting work as expected now.
Thank you for the quick fix.
[image]
[image]

khumarahn, 2020-10-18:
@meteyou While you are at it, is it hard to make the chamber heater appear after all other heaters?

meteyou, 2020-10-18:
@khumarahn heaters are sorted alphabetically. only the temperature_fans come afterwards...

khumarahn, 2020-10-18:
ok, so I can name my heater zzz_chamber_heater and it will be in the end.

Thanks!

khumarahn, 2020-10-18:
Double checked after the print finished, it works. Many thanks!!!!! and I am closing this issue.

commit 3670d4cc (2020-10-04): bugfix: add heater_generic to tools (fix #87)
closed 2020-10-18 by khumarahn