mainsail-crew/mainsail #88: [FR] Selecting tool to extrude

khumarahn, 2020-10-01:
Hi, I think web interface should have a way to select an extruder when extruding filament.

meteyou, 2020-10-04:
what do you think is better. to activate the extruder in the heater list (like DWC) or put this into the extruder control panel? i would implement it with the gcode command:

`ACTIVATE_EXTRUDER EXTRUDER=<config_name>: In a printer with multiple extruders this command is used to change the active extruder.`

khumarahn, 2020-10-04:
For me, the more convenient would be a control near the extrusion panel, because in Klipper active extrude only relates to extrusion; in duet things are more complicated with active and standby temperatures and tool offsets.

bung69, 2020-12-12:
Maybe don't hard code to  ACTIVATE_EXTRUDER as without overriding this command with gcode macros it cant handle tool changing.  Allowing T0 T1  or however the user sets up there tool change macros may be a good idea.

closed 2022-04-24 by meteyou