mainsail-crew/mainsail #89: Last modified time is actually showing the file creation time

tionebrr, 2020-10-02:
Well all is in the title.
Using a RPI3B+, everything up-to-date.

View from windows (via samba):
[image]
View from Mainsail:
[image]

meteyou, 2020-10-02:
without logfiles, there is nothing I can do...

tionebrr, 2020-10-02:
Sure, here they are:
[klippy.log](https://github.com/meteyou/mainsail/files/5317758/klippy.log)
[moonraker.log](https://github.com/meteyou/mainsail/files/5317759/moonraker.log)

theopensourcerer, 2020-10-02:
@meteyou I think he's right. I ssh'ed into my RPi and modified one of the gcode files using nano. Then I refreshed the browser:

[image]

theopensourcerer, 2020-10-02:
Sorry mainsail version: v0.2.2 - v0.8.0-765-g1f3a160f

meteyou, 2020-10-05:
ok. i think moonraker extract the wrong date from the files. i redirect it to Arksine...

tionebrr, 2020-10-06:
Thanks @meteyou

meteyou, 2020-10-15:
should be fixed in moonraker...

closed 2020-10-15 by meteyou

tionebrr, 2020-10-18:
Hi, just updated and it's looking good. Thanks a lot.