mainsail-crew/mainsail #93: Button HOSTREBOOT and HOSTSHUTDOWN don't work

anjoku, 2020-10-04:
Hello

I installed Mainsail on an Orangepi Lite. I used the script from https://github.com/th33xitus/kiauh.
It works so far, only the HOSTREBOOT and HOSTSHUTDOWN buttons don't work for me.

Where can I set what happens when these buttons are pressed?
Example: HOSTREBOOT should execute "sudo shutdown -h now".

how can I create and integrate my own macros?

Thank you Andre

[moonraker.log](https://github.com/meteyou/mainsail/files/5323184/moonraker.log)
[klippy.log](https://github.com/meteyou/mainsail/files/5323187/klippy.log)

meteyou, 2020-10-04:
this is a function from moonraker. mainsail only send a request to moonraker, i only find the request in the moonraker.log, but there is error or something else in the logfile. i think its a right problem. please ask arksine in the moonraker repository.

https://github.com/Arksine/moonraker/blob/master/docs/web_api.md#machine-commands

anjoku, 2020-10-05:
Hello,

thanks for the food for thought ....

I then looked again. I was a little wrong. Host shutdown works. Hostreboot not.

I'll check with Moonraker.

Thank you very much
Andre

closed 2020-10-05 by meteyou