mainsail-crew/mainsail #95: Send files with duetRRF plugin from cura to Mainsail

Matou25, 2020-10-06:
I previously used dwc2-for-klipper-socket.
with duetRRF pluging it's possible to send gcode file or start print directly from cura

plugin is always configured but now i have an error:

2020-10-07 00:33:29,775 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [170]: Matou Kossel-upload | Filename set to: KM_delta_bed_test.gcode
2020-10-07 00:33:29,776 - DEBUG - [MainThread] SliceInfoPlugin.SliceInfo._onWriteStarted [109]: 'info/send_slice_info' is turned off.
2020-10-07 00:33:29,777 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [183]: Matou Kossel-upload | Loading gcode...
2020-10-07 00:33:29,932 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onFilenameAccepted [194]: Matou Kossel-upload | Connecting...
2020-10-07 00:33:29,980 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice.onUploadReady [219]: Matou Kossel-upload | Uploading...
2020-10-07 00:33:30,156 - ERROR - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._onNetworkError [469]: _onNetworkError: 202
2020-10-07 00:33:30,158 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._cleanupRequest [446]: _cleanupRequest called
2020-10-07 00:33:30,160 - DEBUG - [MainThread] DuetRRFPlugin.DuetRRFOutputDevice._cleanupRequest [454]: _cleanupRequest finished

[image]

do you thing it's possible add this fucntionnaly ?

BR

meteyou, 2020-10-07:
you cannot send files to mainsail, only to Moonraker. Moonraker emulate the octoprint api, so you can upload it with octoprint as host. I only think cura does not work. pls test it, otherwise crate a issue in the moonraker project!

Matou25, 2020-10-07:
Thanks  i test cura connexion to moonraker with api key but it won't connect, i will open an issue in moonracker project.

closed 2020-10-07 by Matou25