mainsail-crew/mainsail #96: Refresh needed for interface settings when psu is not always on.

MarcPot, 2020-10-08:
**Describe the bug:**\
Because my PSU is connected to a relay, Klipper is never connected on startup, this causes the mainsail interface settings to not load when I load the page.
After I do a restart so klipper can connect to my MCU I need to refresh the page to load the interface settings.

**Expected behavior:**\
I expect the interface settings to load even when Klipper can't connect.

**Screenshots:**\
Screenshot of what I see before the refresh.
[image]
Screenshot of what I see after the refresh.
[image]

**Logfiles:**\
Please upload your Logfiles here or upload them to [pastebin.com](http://pastebin.com) and paste your log links here.
[moonraker.log](https://github.com/meteyou/mainsail/files/5350929/moonraker.log)
[klippy.log](https://github.com/meteyou/mainsail/files/5350930/klippy.log)

meteyou, 2020-10-10:
this is a problem of V0.2.2. should be fixed in V0.2.3. pls test it again with the next version. but i think we talked about it on discord already 2 times about this issue...

MarcPot, 2020-10-10:
Yea but I opened this issue because I found the cause. And on Discord I only had the problem. Thanks anyways for fixing it!

meteyou, 2020-10-24:
@MarcPot can we close this issue?

MarcPot, 2020-10-25:
@meteyou I haven’t been able to test it because of personal problems. But if you day it’s fixed it’s fixed.

closed 2020-10-25 by MarcPot