mainsail-crew/mainsail #97: [FR]

MarcPot, 2020-10-08:
**Is your feature request related to a problem? Please describe.**
Because my PSU is connected to a relay that's normally open, Klipper can't connect to my MCU on startup. Mainsail tries to connect Klipper anyways and this causes that I have to wait until it times out before I can do anything.

**Describe the solution you'd like**
Or something that detects my psu being off, and doesn't make Klipper auto-connect because of that, giving me the chance to first turn it on.

Or an option to disable to auto-connect.

meteyou, 2020-10-10:
i will test something this weekend, but i think this is a klipper problem and not a mainsail problem. mainsail can only support klipper functions...

LazaroFilm, 2020-10-22:
I have my MCU on a separate PSU as well, th33xitus on Discord posted a method for your pi to detect the connection and attempt a reconnection when your MCU is powered: 
get your vendor and product id of your mcu. -> `lsusb`
then create a udev rule:
```
cd /etc/udev/rules.d/
sudo nano 98-klipper.rules
```

and put in:
```
## rule to restart klipper when the printer is connected via usb
SUBSYSTEM=="usb", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="abcd", ACTION=="add", RUN+="/bin/sh -c '/bin/echo RESTART > /tmp/printer'"
```

this exact line did not work for me (but maybe it does for you), i had to modify it that way:
```
## "/bin/echo RESTART > /tmp/printer" not working, so using the symlink destination instead -> works!
SUBSYSTEM=="usb", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="abcd", ACTION=="add", RUN+="/bin/sh -c '/bin/echo RESTART > /dev/pts/0'"
```

remember to put in YOUR vendor and product id. the ones you see above probably won't work for you

meteyou, 2020-10-24:
Thx for this guide. maybe i should add this to the docs?

closed 2020-10-31 by meteyou

LazaroFilm, 2020-11-01:
> Thx for this guide. maybe i should add this to the docs?

That would be awesome!

Loreagle, 2021-01-10:
> I have my MCU on a separate PSU as well, [th33xitus on Discord](https://discord.com/channels/431557959978450984/550724465886101514/729330554796900433) posted a method for your pi to detect the connection and attempt a reconnection when your MCU is powered:
> get your vendor and product id of your mcu. -> `lsusb`(unplug your MCU, do `lsusb` then plug it in and power up and do another `lsusb` the extra line is your MCU info)
> 
> then create a `udev` rule:
> 
> ```
> cd /etc/udev/rules.d/
> sudo nano 98-klipper.rules
> ```
> 
> and put in:
> 
> ```
> ## rule to restart klipper when the printer is connected via usb
> SUBSYSTEM=="usb", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="abcd", ACTION=="add", RUN+="/bin/sh -c '/bin/echo RESTART > /tmp/printer'"
> ```
> 
> this exact line did not work for me (but maybe it does for you), i had to modify it that way:
> 
> ```
> ## "/bin/echo RESTART > /tmp/printer" not working, so using the symlink destination instead -> works!
> SUBSYSTEM=="usb", ATTRS{idVendor}=="2341", ATTRS{idProduct}=="abcd", ACTION=="add", RUN+="/bin/sh -c '/bin/echo RESTART > /dev/pts/0'"
> ```
> 
> remember to put in YOUR vendor and product id. the ones you see above probably won't work for you

Do you know if there is a way to do this also when my board is connected trough UART and not USB?