mainsail-crew/mainsail #103: [FR] Telegram status messages

Hywelmartin, 2020-10-17:
Send messages when status changes.. and possibility to interact for example quit a failed print

Like Pause and finished like the octoprint plugin

https://core.telegram.org/bots
https://plugins.octoprint.org/plugins/telegram/

meteyou, 2020-10-17:
thx for your feature request, but mainsail is only running in your browser. something like this have to be executing on your host (raspberry). pls make a feature request on moonraker or klipper.

closed 2020-10-17 by meteyou

Hywelmartin, 2020-10-18:
Taught that there was a webserver involved..  that could take care of this kind of things..

thanks for a quick answer..

meteyou, 2020-10-18:
Webserver ist only for serve the static files. rest of the magic comes from Moonraker and klipper.

meteyou, 2020-10-25:
one guy make a telegram bot. this bot use the API of moonraker. https://github.com/Raabi91/moonraker-telegram

Hywelmartin, 2020-10-26:
Thanks try that out
/Martin