mainsail-crew/mainsail #104: Page refresh

fargonaz, 2020-10-19:
Page refresh leaves: Moonraker can't connect to Klippy!
Please check if the Klipper service is running and an UDS (Unix Domain Socket) is configured.

[moonraker (1).log](https://github.com/meteyou/mainsail/files/5404125/moonraker.1.log)

Just udpated to 0.2.4 and applied all the edit to the nginx files as described int he documentation.

I have to separate installs both behave as above.

meteyou, 2020-10-20:
i think, you have a old moonraker version installed with the new mainsail

```
2020-10-19 11:58:17,975 [websockets.py:dispatch()] - Websocket Request::{"jsonrpc":"2.0","method":"server.info","params":{},"id":1510}
2020-10-19 11:58:17,975 [websockets.py:dispatch()] - Websocket Response::{"jsonrpc": "2.0", "error": {"code": -32601, "message": "Method not found"}, "id": 1510}
```

server.info is a new function of moonraker. pls can you update moonraker and test it again?

fargonaz, 2020-10-20:
Gah, I could have bet money on that.  Thanks for the reply.

meteyou, 2020-10-20:
pls test it and if this was the problem, pls close this issue

fargonaz, 2020-10-20:
All is working fine on both installations after installing the latest moonraker.  Thank you.

closed 2020-10-20 by fargonaz