mainsail-crew/mainsail #105: [FR] Home XY button

LazaroFilm, 2020-10-22:
**Is your feature request related to a problem? Please describe.**
My printer needs to home XY first then home Z separately (with my Piezo Orion) I sometimes need to home XY first to bring the nozzle to the front to clean it before probing. Probing X then Y separately pushes my bed down twice due to my safe Z homing routine while a `G28 X0 Y0` only moves the bed down once. 

**Describe the solution you'd like**
Adding a Home XY button next to Home All, maybe with a toggle option in the settings if others don't want it. 

**Describe alternatives you've considered**
I made a Macro for it, but it's buried in all my macro buttons instead of being easily accessible up top

**Additional context**
Add any other context or screenshots about the feature request here.

meteyou, 2020-10-24:
i think it should be possible to implement all this with `save_z_home`.

[https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L310](https://github.com/KevinOConnor/klipper/blob/master/config/example-extras.cfg#L310)

meteyou, 2020-10-27:
@LazaroFilm do you check the config option and does it work for you?

closed 2020-11-05 by meteyou