mainsail-crew/mainsail #106: View panel on left is dark and no item can be selected

michelluccote, 2020-10-24:
I installed klipper/moonraker and now mainsail. The mainsail is able to connect using moon raker to my printer and can control my printer. When it starts it open on the dashboard. I can see the other views but they are all dark and I cannot select any of them.

I would expect to simply select these and go into >_concolse or Heigth Map or G-Code files or settings but none of these can be selected. They are all very light in color and not functional.

My log file is filled with this line:

2020/10/23 16:47:10 [error] 576#576: *85 connect() failed (111: Connection refused) while connecting to upstream, client: fe80::21b:63ff:febc:96e7, server: _, request: "GET /websocket HTTP/1.1", upstream: "http://127.0.0.1:7125/websocket", host: "raspberrypi.local"

But little else. I see nothing out of the ordinary in the klippy.log or the moonraker.log.

meteyou, 2020-10-24:
which log is filled with this line? can you upload your moonraker.log and upload a screenshot or a video what do you exectly see?

michelluccote, 2020-10-26:
First time I put an attachment. Hopefully this works. I recently restarted my mainsail and still can't select any of the items on the left as indicated. 

[https://pastebin.com/HDZtP7Yn](url)

michelluccote, 2020-10-26:
Here is a picture of my mainsail page with the items disabled on the left. The buttons on the displayed page appear to work correctly but no access to other pages of the mainsail UI.

[image]

michelluccote, 2020-10-26:
I had to make some changes to get KlipperScreen to work and some of these changes seem to have fixed my problem with mainsail. Changes I did include:

1. Switch PI to boot to console no login
2. sudo gpasswd -a pi tty
3. sudo apt-get install xserver-xorg-legacy
4. Edited my /etc/X11/Xwrapper.config and but in the line:
     needs_root_rights=yes
     allowed_users=anybody

Not sure which of these has resolved my problem but now I have access to my Console/Heightmap/G-code Files/Settings

meteyou, 2020-10-27:
its look like a browser compactibility problem... do you have update your browser?

michelluccote, 2020-10-27:
I'm using an old Mac in my garage for this work. It's an iMac from 2007 running chrome browser version 86.0.4240.111 (Official Build) (x86_64). I've not changed the browser or the iMac. Only the changes mentioned above made a difference and got it working.

closed 2020-11-05 by meteyou