mainsail-crew/mainsail #113: Mainsail displays wrong object height in gcode manager

toskium, 2020-10-27:
**Describe the bug:**\
SuperSlicer shows 21.80mm object height
Mainsail gcode manager shows 22.20mm object height

**Expected behavior:**\
Identical object height slicer/mainsail

**Screenshots:**\
[image]
[image]

**Buildplate as 3MF**\
[buildplate_3mf.zip](https://github.com/meteyou/mainsail/files/5446476/buildplate_3mf.zip)

**Logfiles:**\
[klippy.zip](https://github.com/meteyou/mainsail/files/5446452/klippy.zip)
[moonraker.zip](https://github.com/meteyou/mainsail/files/5446453/moonraker.zip)

meteyou, 2020-10-27:
all gcode metadata is extracted by moonraker. there was an issue before and a description how this is done and therefore there can be "imprecisions". Arksine/moonraker#21

toskium, 2020-10-27:
Ah, sorry to have bugged you with that. 
I added the fix suggested by Arksine and now it displays the correct object height.

closed 2020-10-27 by toskium