mainsail-crew/mainsail #114: [FR] Emergency Stop Confirmation on Mobile Site

toskium, 2020-10-27:
**Is your feature request related to a problem? Please describe.**
On Android Firefox and Chrome (smartphone) the browser tab button is exactly above the emergency stop button and (at least with my fat fingers) there is the not so slight chance I hit that accidentally when instead I only wanted to switch a browser tab.

**Describe the solution you'd like**
It would be great if there was a config option that would allow to set a confirmation for the emergency button. 
Since it only bugs me on my mobile phone an option for the mobile layout would suffice from my viewpoint.

**Describe alternatives you've considered**
Move my swipe finger more carefully :-)

meteyou, 2020-10-27:
i have already considered this and it is on my todo list. but i want to make this confirm popup optional as some users prefer to have it without a confirmation window.

randellhodges, 2020-10-28:
I'm taking a look at this. I'd like to add options to hide it completely, as well as an option to confirm it.

randellhodges, 2020-10-28:
How about this? Ignore the "do stop" alert, that is just for dev purposes.
[image]

meteyou, 2020-10-30:
@randellhodges i like the idea! is it also possible to use the same button twice? that the button changes to "really stop?" and the second "click" on the button really turns it off? if this works well, i might not even turn the confirm process on and off at all, because clicking twice on the same position is not really "slower" to turn off the printer in an emergency.

randellhodges, 2020-10-30:
I'll take a look at a double press. 

The only (minor) issues I can see with that approach are 1) how long to allow between "double" presses and 2) how to communicate to the user that 2 presses are needed for an emergency. Neither one is a showstopper.

One potential way to solve both. Have 2 buttons, 1 is hidden. When you press the first, it hides itself and shows the 2nd button (in the same spot, but looks different or has some different text). When you click that button, it does the shutdown. If you click *anywhere* else, or maybe set a timer for 5 seconds, then it hides and the first button is visible again. This isn't as fast as a double click but would not require any movement of the mouse or finger beyond clicking the same spot a 2nd time for the stop to happen.

Does that sound reasonable, or too complicated?

meteyou, 2020-11-05:
sorry for the late answer...

i don't care personally, if this is 1 button that changes with a question in it or 2 different buttons that are at the same position. the user will not really notice if they are the same or 2 different buttons. my primary wish would be that it is at the same position and can be executed with 2 clicks at the same position.

triumphtotty, 2021-07-02:
Would it be better to have an option to lock the whole UI?  I've accidentally changed the extrusion slider (to 184%!) when scrolling on mobile.  I now tend to go to the /cam link on mobile first, and then to Dashboard (carefully) from there, but either a "lock" switch for the UI or perhaps a separate page, e.g. /mon with the same info but no controls would be really useful.