mainsail-crew/mainsail #115: Mainsail forgets "Printer Name" after reboot

pheonix3330, 2020-10-28:
**Describe the bug:**\
"Printer Name" does not save when I turn my printer off. It resets to the default "raspberrypi" name instead of the name I set ("Voron V1.8").

**Expected behavior:**\
Mainsail remembers the printer name I set and restores it after reboot. This may be happening because I am cutting power to the MCU when I shut down the printer but I have the raspberry pi powered separately and the pi never turns off. When the printer turns off.

randellhodges, 2020-10-28:
Are you pressing the "save" checkbox after making changes inside settings->interface? After you save, check the gui.json file and ensure it has the correct value in it?

pheonix3330, 2020-10-28:
wow, didnt even know that button was there. I will do that now. Thanks for your help

closed 2020-10-28 by pheonix3330