mainsail-crew/mainsail #117: WebUI Interface Settings

yannvautrin, 2020-11-04:
**Describe the bug:**\
Everytime I restart the WebUI, I have to re-enter the settings from the side panel to edit the webcam parameters...

**Expected behavior:**\
Would like an option to save interface settings

Thanks!

meteyou, 2020-11-04:
on the right bottom, there is a save button...

yannvautrin, 2020-11-08:
Thanks, sorry for being a noob.... I found it!

meteyou, 2020-11-08:
I have to modify it... auto save or a better visibility action...

zellneralex, 2020-11-15:
How about a pop up question if you leave that page and have unsaved changes?

meteyou, 2020-11-15:
I don't like popups 😅... At the beginning of moonraker, it was only possible to upload files, when the printer is in idle mode. So we need the safe button, that the user can save the config after a print or something else. Now we can upload every time and I will directly upload the new settings for every button/switch...

meteyou, 2020-11-27:
i implemented a autosave function

closed 2020-11-27 by meteyou