mainsail-crew/mainsail #118: WebUI unresponsive if opened from multiple clients

toskium, 2020-11-04:
**Describe the bug:**\
I am using multiple computers. One besides the printer, one in my office upstairs.
Usually, on the computer near the printer, I leave the mainsail browser tab open and walk away.

When I open mainsail in a new browser tab on my office computer everything is working fine there.
As soon as the print is finished, I go downstairs to the printer to start another print, this is when mainsail is unresponsive in in that browser window.
The only way to resolve that issue is to close the browser tab, reload mainsail, everything is fine again.

It seems to occur more often when browser tabs are open for several hours, I can't reproduce it right away.

I am using Google Chrome on both computers in the latest stable version.
OS is Windows 10.

**Expected behavior:**\
Multiple mainsail connections working fine, no extra reload/close tab etc.

**Logfiles:**\
[logfiles.zip](https://github.com/meteyou/mainsail/files/5490039/logfiles.zip)

translucentfocus, 2020-11-13:
Hmmm, I'm no expert, but... hopefully (**_if_** I'm on the right track) this saves someone else some time:

- It looks like in the logs your first client 192.168.10.10 timed out
- And [it looks like moonraker has a 60 minute timeout](https://github.com/Arksine/moonraker/blob/697be6a8226c781b5aa900b96931b830b9f91808/moonraker/authorization.py#L17).
- Then you logged in with client 192.168.10.100 (which worked fine?)
- Then you used client 192.168.10.10 hours later, and it's had already been removed (due to the 60 minute timeout) so you had to restart the page

<pre>
<b>2020-11-04 04:29:28,623 [authorization.py:_check_trusted_connection()] - Trusted Connection Detected, IP: 192.168.10.10</b>
2020-11-04 04:29:28,624 [web.py:log_request()] - 101 GET /websocket (192.168.10.10) 1.94ms
2020-11-04 04:29:28,624 [websockets.py:add_websocket()] - New Websocket Added: 139965320175800
...
<b>2020-11-04 05:32:34,467 [authorization.py:_prune_conn_handler()] - Trusted Connection Expired, IP: 192.168.10.10</b>
2020-11-04 08:34:47,041 [websockets.py:remove_websocket()] - Websocket Removed: 139965320175800
2020-11-04 19:00:32,978 [websockets.py:dispatch()] - Websocket Request::{"jsonrpc":"2.0","method":"printer.gcode.script","params":{"script":"G28"},"id":868}
2020-11-04 19:00:44,349 [websockets.py:dispatch()] - Websocket Response::{"jsonrpc": "2.0", "result": "ok", "id": 868}
2020-11-04 19:00:53,659 [websockets.py:dispatch()] - Websocket Request::{"jsonrpc":"2.0","method":"printer.gcode.script","params":{"script":"preheat_abs"},"id":2147}
2020-11-04 19:00:53,661 [websockets.py:dispatch()] - Websocket Response::{"jsonrpc": "2.0", "result": "ok", "id": 2147}
2020-11-04 19:03:54,697 [websockets.py:remove_websocket()] - Websocket Removed: 139965320302720
<b>2020-11-04 19:04:00,549 [authorization.py:_check_trusted_connection()] - Trusted Connection Detected, IP: 192.168.10.100</b>
2020-11-04 19:04:00,550 [web.py:log_request()] - 101 GET /websocket (192.168.10.100) 1.50ms
<b>2020-11-04 19:04:00,550 [websockets.py:add_websocket()] - New Websocket Added: 139965320189488</b>
...
<b>2020-11-04 19:33:42,440 [authorization.py:_check_trusted_connection()] - Trusted Connection Detected, IP: 192.168.10.10</b>
2020-11-04 19:33:42,441 [web.py:log_request()] - 200 GET /api/version (192.168.10.10) 2.08ms
2020-11-04 19:33:42,758 [file_manager.py:_update_file_list()] - Updating File List <gcodes>...
2020-11-04 19:33:42,759 [web.py:log_request()] - 200 POST /api/files/local (192.168.10.10) 276.51ms
2020-11-04 19:33:42,864 [shell_command.py:run()] - Command (/home/toskala/moonraker-env/bin/python /home/toskala/moonraker/moonraker/plugins/../../scripts/extract_metadata.py -p /home/toskala/sdcard -f 'Filament_Card.gcode') finished
<b>2020-11-04 19:33:53,089 [websockets.py:remove_websocket()] - Websocket Removed: 139965320189488</b>
2020-11-04 19:34:03,366 [web.py:log_request()] - 101 GET /websocket (192.168.10.100) 1.51ms
2020-11-04 19:34:03,366 [websockets.py:add_websocket()] - New Websocket Added: 139965320190104
</pre>

meteyou, 2020-11-13:
it could be that it was also a problem with the old subscripte system of moonraker. it could be that it will be better with the new version. from now on every client has its own subscription. but i'm not finished with the conversion for that. i will test it for all cases.

toskium, 2020-11-13:
@translucentfocus thanks for analyzing the log, that is exactly my type of usage scenario.

@meteyou okidoki, I'll update this issue once you released a new version and hopefully close it then.

meteyou, 2020-11-28:
@toskium pls update and test it again...

toskium, 2020-11-28:
I updated to the latest version of mainsail, unfortunately the issue still exists.
I am not sure if this is connected but since 0.3.0 the console stops working after some time. I noticed while doing a  bed_mesh_calibrate method=manual. Since it takes relatively long and temperature feedback scrolls through the console I suspect it renders the browser tab unusable after a while.
It can be resolved by closing the tab and reloading mainsail.
I will open a separate issue on this tomorrow.

meteyou, 2020-11-28:
@toskium you have update Moonraker also?

toskium, 2020-11-29:
@meteyou ah, not yet. Will do today and provide feedback!

toskium, 2020-12-01:
I finally had some time testing this. As you suggested, I updated moonraker to the latest version but without success.
The original issue remains the same.
Here's the current klippy.log and moonraker.log.
[logs.zip](https://github.com/meteyou/mainsail/files/5622374/logs.zip)

meteyou, 2020-12-01:
you logfile is full of this:

```
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/toskala/moonraker/moonraker/moonraker.py", line 212, in _execute_method
    ret = self.remote_methods[method_name](**kwargs)
  File "/home/toskala/moonraker/moonraker/moonraker.py", line 362, in _process_status_update
    conn.send_status(conn_status)
  File "/home/toskala/moonraker/moonraker/websockets.py", line 312, in send_status
    self.websockets.pop(self.uid, None)
AttributeError: 'WebSocket' object has no attribute 'websockets'
```

do you update the moonraker dependence?
[https://meteyou.github.io/mainsail/update/moonraker#update-moonraker-dependence](https://meteyou.github.io/mainsail/update/moonraker#update-moonraker-dependence)

toskium, 2020-12-02:
Updating moonraker to the latest version and also updating moonraker's dependencies solved the issue.
Thank you!

closed 2020-12-02 by toskium