mainsail-crew/mainsail #119: [FR] Make printer.cfg editable in panel in dashboard.

r-franke, 2020-11-05:
**Is your feature request related to a problem? Please describe.**
Often when I'm writing macro's I have to check coordinates and such. I have to either open up 2 windows to be able to edit the `printer.cfg` and look at the coordinates at the same time.

**Describe the solution you'd like**
Create an (optional) panel for the dashboard where we can edit the `printer.cfg`.

Maybe something like this:
[image]

**Describe alternatives you've considered**
Another option would be to display a statusbar always on top with coordinates and temperature so that it's always visible in other screens.

mattmon, 2020-12-31:
I too would appreciate a similar feature :+1:
 
This would ease switching between the separate configurations for each different tool head used with my TAZ 6.

Here's my take on @r-franke's suggestion:
[image]

eliteSchwein, 2021-01-09:
Since i use alot printer.cfg and already changed alot in mainsail on my repo i will add a configuration panel for printer.cfg and will make a push request

eliteSchwein, 2021-01-10:
Sorry i read the issue wrongly, i expected that mainsail doesnt have a printer.cfg editor, but it has so yeah i dont change anything at that part lol

closed 2021-04-20 by meteyou