mainsail-crew/mainsail #121: [FR] Job history

MrEd88, 2020-11-05:
**Is your feature request related to a problem? Please describe.**
Keep tracking of all printed parts is a bit of work, we always have to consider which filament, settings and time it takes to print a piece. Print time estimation is always inaccurate for any gcode analyzer (online services, slicers, mainsail)

**Describe the solution you'd like**
Keep a log of all print jobs done to track real used time to really know how much does it takes  to print a piece, it will be useful too to compare to slicers settings and tune accordingly to get better estimations.

**Describe alternatives you've considered**
Keeping an spreadsheet logging the time is the only alternative I have now. Previously I used one of the available octoprint plugin.

**Additional context**
As a sidenote, is there any working on plugin support?

meteyou, 2020-11-05:
mainsail is only a browser app. so if you don't have your browser open and a print is complete, you can't log anything. mainsail only shows the current klipper status and doesn't do anything by itself. the logging would have to be on the klipper/host side (maybe in the virtual_sdcard). how far i can use this log on the client side or if the calculation has to be done on the klipper/host side, we would see than. feel free to make a PR in klipper for this feature.

an approach to get a better remaining time would be this option: KevinOConner/klipper 3303

plugins themselves would have to be installed on the host & on the client side to be able to implement something halfway useful. furthermore plugins are not planned at the moment. stability and full functionality of klipper is currently the main priority.

meteyou, 2021-03-11:
with this moonraker PR, it will be possible to integrate it -> https://github.com/Arksine/moonraker/pull/109

meteyou, 2021-03-15:
pls update moonraker and add [history] in your moonraker.conf to start collecting data. in the next release, there will be a print history. here a small preview:

[image]

itCarl, 2021-03-17:
@meteyou 
would it be possible to somehow differentiate beetwen different FIlaments not just by type [PLA, PETG, ABS; etc.]?
I'm thinking not just about Statistics but a whole Filament stock manager or something.

**So a possible implementation would look like following in my Head:**
a Slicer plugin that puts some information about the selected filament name, color, brand, and more specs for example.
This way "Moonraker" could get and save more data about a printed object.
**current cura gcode:**
[image]
**My Idea would be some thing like:**
[image]

**OR**
(another thing that just came to my mind while i wrote these stupid things above.)

Would it be possible to just use the Monraker [rest api database endpoint](https://github.com/Arksine/moonraker/blob/master/docs/web_api.md#database-apis) to integrate some system like i tried to explane above? But just without that whole slicer Plugin thingy... just a Mainsail/Moonraker "Plug-in" where a user can add a Filament to a past print job and or a current Print.
So that there is a Statistic for each spezific Filament .

meteyou, 2021-03-17:
at first, there will be only a print statistic. if it works fine, there will come a spool tracking system, but something like this will be implemented in some way...

ayufan, 2021-04-09:
@meteyou Would you mind pushing a new release and publishing it http://my.mainsail.xyz/?

meteyou, 2021-04-20:
published in v1.5.0

closed 2021-04-20 by meteyou